u-root/u-root #2376: gosh is significantly larger than rush

hugelgupf, 2022-04-06:
**Describe the bug**
gosh is significantly larger than rush. In our use cases, when replacing rush with gosh, gosh increases compressed stripped initramfs sizes by 5%.

**To Reproduce**
Steps to reproduce the behavior:

```
$ git checkout 70fe326c5cce701b27ae2ec8d0093818af911b22 # commit before rush was deleted
$ go build
$ ./u-root core cmds/exp/gosh
18:09:52 Successfully built "/tmp/initramfs.linux_amd64.cpio" (size 13213204).
$ ./u-root core ./cmds/exp/rush
18:10:07 Successfully built "/tmp/initramfs.linux_amd64.cpio" (size 12594708).
```

hugelgupf, 2022-04-06:
```
-rwxr-x---  1 chrisko primarygroup 2.4M Apr  5 18:11 rush
-rwxr-x---  1 chrisko primarygroup 3.8M Apr  5 18:12 gosh
```

hugelgupf, 2022-04-06:
https://github.com/u-root/u-root/pull/2377 turns the test back on that probes for dead code elimination. gosh seems to enable dead code elimination in Go, so that's not the issue.

MDr164, 2022-04-06:
@rminnich told adding a few extra KB isn't too bad if that means maintaining less code and having added features like full bash script support. But looks like those extra KB are causing you troubles now? The compressed initramfs are indeed different in size and removing the tab completion feature from gosh (which is still slightly broken when running without a framebuffer) scraped off some of the extra space:
```
3.5M	./initramfs.gosh.cpio.xz
3.3M	./initramfs.rush.cpio.xz
3.4M	./initramfs.gosh_notabcomp.cpio.xz
```

Should we bring rush back which is less capable of parsing shell code than gosh and has more code to maintain or should we fix and slim down tab completion and except the few added KB of the initramfs? Or is the last option not feasible?

hugelgupf, 2022-04-08:
can you quantify down to the kb rather than rounding to 3.3, 3.4 and 3.5MB? on some platforms, they could cause us trouble.

It could be worth it to have a gosh without tab completion

MDr164, 2022-04-11:
```
3500	./initramfs_gosh.cpio.xz
3472	./initramfs_gosh_notabcomp.cpio.xz
3320	./initramfs_rush.cpio.xz
```

As discussed with Ron I'll look into tab completion again and try to remove it's current shortcomings so that we can use it. This will likely mean reimplementing basic tab completion in u-root instead of using a 3rd party library.

closed 2024-03-04 by hugelgupf