u-root/u-root #2387: cmds/core/wget should support -O to stdout

floren, 2022-04-28:
**Is your feature request related to a problem? Please describe.**
There's no good way to fetch something from HTTP to stdout.

**Describe the solution you'd like**
`wget -O - http://google.com` would work.

**Describe alternatives you've considered**
It might be good to have a `curl` command too, or for those of us on Plan 9, `hget`.

floren, 2022-06-05:
Note that you can't just say `wget -O /fd/1 http://example.org` on Plan 9:

```
wget: 2022/06/05 15:19:50 Failed to create output file "/fd/1": open /fd/1: '/fd/1' inappropriate use of fd
```

rminnich, 2022-06-08:
that sounds like an error in how wget opens the file I bet.

rminnich, 2022-06-08:
return OpenFile(name, O_RDWR|O_CREATE|O_TRUNC, 0666)

yep. 

more unix creeps in ...

rminnich, 2022-06-08:
oh shit.
        case Qfd:
                if(omode != OREAD)
                        error(Eperm);
                break;

so this won't work anyway. Damn.

closed 2024-04-10 by binjip978