u-root/u-root #2427: test-integration-arm flaky

hugelgupf, 2022-06-15:
test-integration-arm has been flaky lately. People have been blaming that on timeouts, but if you look at the actual tests, that's not the case.

```
    integration.go:145: 2022/06/15 18:22:26 serial: Kernel panic - not syncing: Attempted to kill init! exitcode=0x0000000b~
```

exit status 0xb is a signal, signal 0xb (= 11) is SIGSEGV.

@rjoleary recommends moving to qemu 7.

cc @MDr164 @10000TB

10000TB, 2022-06-15:
Making an attempt to upgrade arm image to 7.0.0 and see if that helps

10000TB, 2022-06-24:
much stable now.  18/19 most recent runs passed https://app.circleci.com/pipelines/github/u-root/u-root?filter=all

10000TB, 2022-06-28:
Close this for now, as from my runs in the past few days it is no longer flaky as reported.

closed 2024-02-20 by hugelgupf