u-root/u-root #2462: Add a mkdir command to the command line

rminnich, 2022-08-03:
**Is your feature request related to a problem? Please describe.**
Sometimes we need an empty directory in the initramfs for tools like cpu. Currently it's a bit inconvenient: one must do something like this:
mkdir empty
u-root -files empty:sym -files empty:export ...

**Describe the solution you'd like**
We might allow a mkdir command
u-root -mkdir empty -mkdir sym
or we might allow a special sentinal value meaning empty directory? What might that be? It can't be /dev/null, that's a file.

**Describe alternatives you've considered**
None

rjoleary, 2022-08-17:
Why not `u-root -files $(mktemp -d):sym -files $(mktemp -d):export`?

rjoleary, 2022-08-17:
I feel we have to be careful with `-mkdir` because the next request will be `-chmod`, then `-chown`, etc.. Until we have reimplemented an entire filesystem shell on the u-root cmdline.

rminnich, 2022-09-21:
you are right.

closed 2022-09-21 by rminnich