u-root/u-root #2487: localboot -grub -d :mounted:[];no boot configuration found

hanxinwu1, 2022-08-28:
**Describe the bug**
When I run u-root localboot on a server with a hard disk and can boot normally, I enter localboot -grub -d on the u-root command line but I can't recognize the hard disk in the system, and when I can directly find the grub configuration on another server and boot into the operating system, is there any problem with localboot?

**u-root command line input
/#localboot -grub -d
/#mounted: []
/#No boot configuration found

**Additional context**
 I really wanna to know why localboot do this ,can you resolve my doubts?Thanks!

hanxinwu1, 2022-08-28:
Is it because of my server's hard drive that it is not recognized?

johnnylinwiwynn, 2022-08-28:
Please try 'boot' command, could be related to
https://github.com/u-root/u-root/issues/2203
Another possibility is that the LinuxBoot kernel is not compiled with the file system types used by your hard drive, such as XFS support., etc.

hanxinwu1, 2022-08-30:
> Please try 'boot' command, could be related to #2203 Another possibility is that the LinuxBoot kernel is not compiled with the file system types used by your hard drive, such as XFS support., etc.
Firstly,boot cmd also outputs mounted:[].
#boot -no-exec -noload -v
2022/10/25 Booting from the following block devices :[]

The server contains the operating system is a sata hard disk, and then it is divided into three partitions, the following is my server's hard disk partition and various file system types.The information about uuid and partuuid in the Log log output in lsblk is omitted and not written here.
/#lsblk
/dev/sda1: TYPE="vfat"
/dev/sda2: TYPE="ext4"
/dev/sda3: TYPE=LVM2_member
/dev/mapper/ubuntu--vg-ubuntu--lv:TYPE="ext4"

my Linux kernel put the ext4 file system compiled into it, theoretically at least can find my hard disk partition file system is ext4 partition such as /dev/sda2 and /dev/mapper/ubuntu--vg-ubuntu--lv, but the mount point found is empty,I just can't understand it.> Please try 'boot' command, could be related to #2203 Another possibility is that the LinuxBoot kernel is not compiled with the file system types used by your hard drive, such as XFS support., etc.
Firstly,boot cmd also outputs mounted:[].
#boot -no-exec -noload -v
2022/10/25 Booting from the following block devices :[]
2022/10/25 
2022/10/25
2022/10/25
The server contains the operating system is a sata hard disk, and then it is divided into three partitions, the following is my server's hard disk partition and various file system types.The information about uuid and partuuid in the Log log output in lsblk is omitted and not written here.
/#lsblk
/dev/sda1: TYPE="vfat"
/dev/sda2: TYPE="ext4"
/dev/sda3: TYPE=LVM2_member
/dev/mapper/ubuntu--vg-ubuntu--lv:TYPE="ext4"

my Linux kernel put the ext4 file system compiled into it, theoretically at least can find my hard disk partition file system is ext4 partition such as /dev/sda2 and /dev/mapper/ubuntu--vg-ubuntu--lv, but the mount point found is empty,I just can't understand it.Are there other possible causes for this?Thanks.

hanxinwu1, 2022-08-30:
> Please try 'boot' command, could be related to #2203 Another possibility is that the LinuxBoot kernel is not compiled with the file system types used by your hard drive, such as XFS support., etc.
This is the file system compiled into my Linux kernel menuconifg, there are only two, as follows.
[image]

johnnylinwiwynn, 2022-08-30:
Looks like your SATA device is not recognized by Linux, /dev/sda* need to be created first. Need to debug from earlier boot firmware and it's not in u-root's scope. Maybe your firmware (UEFI PEI or coreboot., etc) doesn't configure for your SATA device correctly.
For our case m.2 nvme drive usually works.

hanxinwu1, 2022-09-05:
> Looks like your SATA device is not recognized by Linux, /dev/sda* need to be created first. Need to debug from earlier boot firmware and it's not in u-root's scope. Maybe your firmware (UEFI PEI or [...]

My server can be booted normally with the original manufacturer's uefi file, which means that the pei stage should be no problem for hard disk initialization, right? Linuxboot is to replace the uefi shell with the Linux kernel, in practice I used the Linux-4.19.6 kernel compilation to replace, do you mean that the existence of the hard disk cannot be recognized in the linux-4.19.6 kernel I compiled? Because of the space of the uefi shell, my Initramfs compilation only contains localboot and fnetboot and {init, dhclient, elvish, mount, ls} these commands, in fact my compilation initramfs commands are as follows: 
u-root -build=bb ./cmds/core/{init, dhclient, elvish, mount, ls } ./cmds/boot/localboot 

This initramfs compilation should not be the cause of my error, right? Is it the linux-4.19.6 compilation menuufig configuration error that causes the hard disk not to be recognized? If I want Linux-4.19.6 to recognize m.2 nvme drives, how do I configure the Linux-4.19.6 menuconfig option?

hanxinwu1, 2022-09-05:
> Looks like your SATA device is not recognized by Linux, /dev/sda* need to be created first. Need to debug from earlier boot firmware and it's not in u-root's scope. Maybe your firmware (UEFI PEI or [...]

The hard disk on my server is u.2 interface to m.2 interface, will this be the reason why the Linux kernel cannot recognize the system hard disk? I only put sata and scsi into Linux kernel.Thanks.

PR #2924 (2024-02-24): Move less-maintained boot commands
`fbnetboot`, `localboot`, and `systemboot` are not well-maintained here upstream. We get questions from people about `localboot` all the time, but as we've merged most `localboot` functionality into `boot` and most current maintainers are more familiar with `boot` and `pxeboot`, let's move them "out" front and recommend those.

Fixes #2911
Fixes #2487
Fixes #2483
Fixes #2142
closed 2024-02-24 by hugelgupf