u-root/u-root #2498: base64 skips first argument and waits for stdin

binjip978, 2022-09-17:
**Describe the bug**
base64 waits for stdin when one parameter was passed. If two parameters were passed it encodes first argument.

**To Reproduce**
Steps to reproduce the behaviour:
```sh
./base64 file.txt
# waits for stdin
./base64 file.txt file.txt
YmFzZTY0LWJ1%
```

**Expected behavior**
Should encode with one argument, like GNU or macOS base64.

**Additional context**
Probably happens because flag.Args return length of non command arguments, with one parameter it length is equal 1 and base64 waits for stdin. The change of > to >= should fix the problem.

rminnich, 2022-09-18:
Thanks, I'm looking at this now.

rminnich, 2022-09-21:
fixed by #2499

10000TB, 2022-09-21:
@binjip978  fix is in, try from HEAD, and let us know !

closed 2022-09-21 by 10000TB