u-root/u-root #2511: pxeboot/fbnetboot:could not configure eth0 for ipv4

hanxinwu1, 2022-10-05:
**Describe the bug**
when I use pxeboot or fbnetboot like this
[image]
The command-line output is “could not configure eth0 for ipv4,unable to open a broadcasting socket: address family not supported by protocol” 
My network card is Intel's and the driver is loaded, do you know what causes it, is this a bug?
Thanks!

hanxinwu1, 2022-10-07:
Could not configure eth0 for IPv4: unable to open a broadcasting socket: address family not supported by protocol.
Is linux kernel's net part causing this problem?And how can I solve it？Can you think of any workarounds？Thank you!

hanxinwu1, 2022-10-07:
And when I use /cmd/boot/fbnetboot,that mean use fbnetboot,I get this log.
[image]
Is this the reason I didn't set up a pxe server? How to set up a good fbnetboot server, the same as pxeboot? What is the difference between pxeboot and fbnetboot?

Keno, 2022-10-10:
Check whether your kernel config has `CONFIG_PACKET` set. I had the same issue and setting that fixed it.

hanxinwu1, 2022-10-11:
> 

Yes, it works. Another question I would like to ask is that when I use dhclient, sometimes I can correctly assign an ipv4 address but sometimes I will report the following errors.
[image]
Do you know why?

Keno, 2022-10-11:
Yes, you're missing entropy. Assuming you're in qemu, you should configure your kernel to support the virtio entropy source and then add that to the qemu launch line.

hanxinwu1, 2022-10-11:
> Yes, you're missing entropy. Assuming you're in qemu, you should configure your kernel to support the virtio entropy source and then add that to the qemu launch line.

I am running in the server results. When I type this command, sometimes it works, sometimes it doesn't work.
Then when it runs successfully, enters the new kernel interface, and starts to download the system, it has an error.
Error :probing for devices to install to failed (sata disks cannot be identified)
[image]

Do you know that's why?(I guess :Is there no sata hard disk drive for the ubuntu operating system ubuntu22.10 downloaded from my website)

hanxinwu1, 2022-10-12:
> 

I tried the pxeboot command to install the system today for another day, and I ran it on a physical server. I tried ubuntu 22.10 again, and I was already on the interface for configuring the language; 
[image]

but the error after that was: probing for devices to install to failed.
[image]

 I also tried to install centos7 or centos8, which had an error before the kernel initialization had started http downloading the iso file, both of which were reported as: no carrier detected on interface eth0.
[image]

 I hadn't done a pxe installation system before that, which was a bit tricky for me.It's a more annoying thing.Keep up the good work tomorrow haha.

closed 2024-03-04 by hugelgupf