u-root/u-root #2517: SIGSEGV in fbnetboot

Keno, 2022-10-10:
**Describe the bug**
I'm playing around with fbnetboot in qemu and encountered a SIGSEGV. I'm probably not doing
anything sensible, but a SIGSEGV seems problematic. The backtrace is:

```
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x40 pc=0x7c9362]

goroutine 1 [running]:
github.com/u-root/u-root/cmds/boot/fbnetboot.boot({0xc000035f9c?, 0x0?}, 0x1101fa0?)
	github.com/u-root/u-root@v0.0.0/cmds/boot/fbnetboot/main.go:243 +0xac2
github.com/u-root/u-root/cmds/boot/fbnetboot.registeredMain()
	github.com/u-root/u-root@v0.0.0/cmds/boot/fbnetboot/main.go:112 +0x6eb
bb.u-root.com/bb/pkg/bbmain.Run({0x7fff63655e53?, 0xc000035ef8?})
	bb.u-root.com/bb/pkg/bbmain/register.go:134 +0x82
main.run()
	bb.u-root.com/bb/main.go:177 +0x4a
main.main()
	bb.u-root.com/bb/main.go:185 +0x65
```

**To Reproduce**
Build u-root master using `u-root core boot` with kernel 5.16 as a coreboot (master) payload, then run `sudo qemu-system-x86_64 -M q35 -enable-kvm -bios build/coreboot.rom -m 1024 -object rng-random,filename=/dev/urandom,id=rng0 -nographic -netdev user,id=user.0 -device e1000,netdev=user.0`

Then at the elvish shell run:
`fbnetboot --skip-dhcp --skip-cert-verify --netboot-url https://jc-rootfs-images.s3.amazonaws.com/xtrxbot/v1/default.pxe`

**Expected behavior**
I'm pretty sure I'm doing the networking wrong, so I didn't expect it to work, but I didn't expect it to segfault either.

abrender, 2022-11-11:
I've poked around in at https://github.com/u-root/u-root/blob/970fddfe138377be47707aeadce1b539b62dcb49/cmds/boot/fbnetboot/main.go#L163 for a while but line 243 doesn't point to anything useful and all the pointers I found are properly handled with nil values, and all error checking that I can see is performed. So I'm stumped.  I tried to reproduce this in qemu and I get the expected error (the `--netboot-url` flag is supposed to be the url to a bzimage/vmlinuz file, not the boot parameters).

@Keno can you please add more context and logs for all of the lines preceding that sigsevg? Hopefully that'll help us to track down where in `boot` that nil pointer dereference is happening.

abrender, 2022-11-11:
I've found one potential cause for nil pointers because of invalid error handling - will be fixed by https://github.com/insomniacslk/dhcp/pull/483 and https://github.com/u-root/u-root/pull/2556. 

However, I don't believe that is what caused your error because you have `--skip-dhcp` and the stack trace is very different.

Keno, 2022-11-11:
Let me check if I still have the image. I posted exactly what I tried, I don't recall there being anything particularly more interesting in the logs - had hoped the backtrace would be sufficient, but let me see if I can dig it up again.

abrender, 2022-11-11:
Okay, I was able to reproduce this :)  It requires that qemu has internet connectivity, but we pass in a URL with a port that is unreachable. For example: `fbnetboot --skip-dhcp --skip-cert-verify --netboot-url https://jc-rootfs-images.s3.amazonaws.com:444/xtrxbot/v1/default.pxe`

After attempt #3 we get the same sigsegv as mentioned in the bug report.

The line causing the sigsegv is `defer resp.Body.Close()` (https://github.com/u-root/u-root/blob/8757e8fd13f47137b9794c49f04baaf268d0e7a2/cmds/boot/fbnetboot/main.go#L256)

The preceding code (https://github.com/u-root/u-root/blob/8757e8fd13f47137b9794c49f04baaf268d0e7a2/cmds/boot/fbnetboot/main.go#L238-L254) has a subtle bug - it will actually continue outside of the loop after maxHTTPAttempts if the error is `retryableNetError` (there's good connectivity) or `retryableHTTPError` (can't reach the port on the server).

I created a simple reproduction here: https://go.dev/play/p/yOIQFuIvRfZ

I am out of time tonight to create a patch, but I can send a patch tomorrow to get this fixed.

Thanks for the bug report!

Keno, 2022-11-11:
Great, sorry for the incomplete reproducer. I was able to find the image, so if I need to try anything else, let me know :).

abrender, 2022-11-11:
> had hoped the backtrace would be sufficient, but let me see if I can dig it up again.

Yeah normally the backtrace is very helpful, but in this case the busybox-like setup recompiles/mashes all the binaries and the line numbers get a bit scrambled, so the line number pointed to the right area, but wrong exact line.   Anyway, the problem is located now, patch will come tomorrow :) Thanks again for finding and reporting this!

abrender, 2022-11-11:
https://github.com/u-root/u-root/commit/94a57100c19663c98c4613b8e296fe0d0d776d43 should fix this issue - I'm waiting for https://github.com/u-root/u-root/pull/2556 to be merged in first so that there are no merge conflicts.

@Keno  Note that while this will fix the sigsegv, it won't actually fix your issue. Note that the `--netboot-url` parameter passed to `fbnetboot` should be the path to the vmlinuz/bzimage file.

closed 2022-11-12 by probot-auto-merge[bot]

hugelgupf, 2022-11-12:
Really I would like to deprecate fbnetboot as much as possible and point to pxeboot as our standard implementation (that's where effort should be going anyway)

hugelgupf, 2022-11-12:
We could add a --netboot-url argument like that to pxeboot for example.