u-root/u-root #2527: cmds/core: provide importable main-like APIs for coreutils

mvdan, 2022-10-16:
**Is your feature request related to a problem? Please describe.**

I've been looking for a reasonably complete implementation of coreutils in Go for some years. The context is https://github.com/mvdan/sh/issues/93; projects like https://github.com/go-task/task use the shell interpreter to make their makefile-like scripts portable to platforms like Mac and Windows. As you can imagine, that tends to fall short when scripts attempt to use coreutils binaries.

**Describe the solution you'd like**

Coreutils implementations, like `cp` and `mv`, which I can import as Go packages and call.

It is important that they be importable as Go APIs, so that they are bundled as part of the same binary and can be called without executing a binary. A new process and an `exec` call are not a problem per se, but that would require setting up a binary to be called. I can't reasonably do that as a library author so it wouldn't work out of the box.

Portability on a best-effort basis is also wanted. It's natural that some shell or coreutils features may not work properly on Windows, for example, but most basic functionality like `cp foo bar` should work fine.

u-root already has fairly complete and portable coreutils, which is a delight to have found out :) For example, [cp is here](https://github.com/u-root/u-root/tree/main/cmds/core/cp), and it's also [available as a library](https://github.com/u-root/u-root/tree/main/pkg/cp). However, the library is not quite what I want - it does not handle the flags and other forms of terminal UI, for example. I want something much closer to a `main` function, such as:

```
func MainRun(args []string, stdin io.Reader, stdout, stderr io.Writer) (exit int)
```

A real `func main` caller would then do:

```
func main() {
    os.Exit(MainRun(os.Args[1:], os.Stdin, os.Stdout, os.Stderr))
}
```

In my case, I would call the function in my interpreter whenever a script calls `cp`, passing in the arguments and stdin/out/err set up in the interpreter. The `MainRun` function would not use any global state, so that I could call it repeatedly - or call `cp.MainRun` after having called `mv.MainRun`, for instance.

This would require a minor incremental refactor for each coreutil program to expose such a MainRun API. I am not set on the name or signature, but hopefully the example helps illustrate what I mean. Here is an example of how I would change `cp` to make this change: https://clbin.com/hvKAB

I'd be happy to make PRs to make these changes incrementally. I would first do it for the handful of coreutils which are most commonly used: `cp`, `mv`, `rm`, `mkdir`, and so on. I don't think we need to refactor all the programs at once. The only open question in my example patch above is what importable package would hold `MainRun`.

**Describe alternatives you've considered**

https://github.com/ericlagergren/go-coreutils looked like the best option some years back, but it went unmaintained.

Established implementations like busybox are tempting, but they are not portable - they do not support Windows, for example. Not being in Go is also a problem, as bundling the project inside a Go library becomes very tricky.

https://github.com/uutils/coreutils is portable and more modern, but being in Rust, it's also very tricky to bundle in a Go library. I considered using `go:embed` with a prebuilt binary, for example, but their busybox-like release binary weighs over three megabytes.

mvdan, 2022-10-16:
I should also note that I'd like every coreutil to use the same `MainRun` signature, so that I could add support for more coreutils via a table like:

```
type bundledMain = func(args []string, stdin io.Reader, stdout, stderr io.Writer) (exit int)

var bundledCoreutils = map[string]bundledMain{
  "cp": cp.MainRun,
  "mv": mv.MainRun,
  "rm": rm.MainRun,
  [...]
}
```

andreynering, 2022-10-19:
Having the ability to import builtins would be hugely helpful for us. Many users are interested in having them available in other tools.

Considering the builtins are already implemented, and only a relatively straightforward refactor is needed to move them to non-main packages as functions, I really hope you accept this feature request.

mvdan, 2022-10-25:
Accepting this might be hard without some experimentation. In that spirit, I'm more than happy to volunteer my time and work on this in a soft fork or a branch in this repository. I could make the changes necessary for a handful of coreutils like `cp` and `mv`, and then that should be enough to tell whether the changes are acceptable or a different design is needed.

Either way, I definitely don't intend to rush the maintainers, but I also don't want to be blocked myself for too long :)

mvdan, 2022-11-12:
I didn't get an answer, which is understandable given we're all busy :) To not keep my project blocked, I'll go ahead with a "soft fork" where I'll maintain the changes I need in the coreutils packages. Always happy to upstream those in the future, of course. I'll keep this issue open for that purpose.

mvdan, 2022-12-15:
Here's my soft fork - right now it only exposes the `cp` implementation: https://github.com/mvdan/sh/tree/93-coreutils

Still very much happy to talk through the design and upstream this. The changes are noisy, but that's simply because half of the `cp` coreutil is currently implemented inside a `package main`.

Also see https://github.com/mvdan/sh/issues/93#issuecomment-1353857765 for how I'm integrating this downstream. And also upstream, actually, given that you use our `interp` package? At least there wouldn't be any package import cycles, but perhaps I should be wary of module dependency cycles :)

hugelgupf, 2022-12-18:
I wouldn't be against something like this. As you can see, we've been transforming a lot of the command's main() files to be more testable, which sort of makes such things happen anyway: https://github.com/u-root/u-root/blob/e4efe471e569b16602b88f53b91a92bb1929680c/cmds/core/cp/cp.go#L121

Though, principally, I wouldn't want to mandate this. The idea of u-root is that you don't have to write all the commands such that someone can include them somewhere else, -- u-root does the including for you with the busybox rewriting magic.

But I'm happy to do it for commands others _want_ to import like this.

Typically, I've pushed for stuff like `cp` to have a `cmds/core/cp` and a `pkg/cp`, where most of the functionality is testable and in `pkg/cp`, and `pkg/cp` has an idiomatic-ish Go API. You actually want to use it for shell builtins though, so you want the real deal with the flags and all.

Any ideas on structure? I don't think a second `pkg/cp` would make sense for us

mvdan, 2023-01-22:
> You actually want to use it for shell builtins though, so you want the real deal with the flags and all.

Correct.

> But I'm happy to do it for commands others _want_ to import like this.

I think that's a reasonable policy. I don't think I will be importing _every_ coreutil program as Go packages, simply because some of them are very rarely used and it's not worth the effort. I'd initially only do the most common dozen or so.

> Any ideas on structure? I don't think a second `pkg/cp` would make sense for us

My thinking for now is to reuse the same `pkg/cp` to expose the new API, for the sake of not creating even more packages. As long as we name the API such that it's clear that it's to mimic the `os/exec` behavior, I don't think it needs a separate package.

I actually didn't link my soft fork properly before. It's here: https://github.com/mvdan/u-root-coreutils/tree/coreutils

I evolved the API slightly due to necessity. `args []string` plus stdin/stdout/stderr isn't enough - `cp` is also affected by the current directory, and a few builtins will also be affected by environment variables. Here's the new API in the soft fork:

```
type RunParams struct {
    Dir string
    Env []string
        
    Stdin          io.Reader
    Stdout, Stderr io.Writer
}

func (p RunParams) MkAbs(name string) string {
    if filepath.IsAbs(name) {
        return name
    }    
    return filepath.Join(p.Dir, name)
}
    
func RunMain(params RunParams, args ...string) (exit int)
```

`RunMain` should likely live in a shared place, as it's not specific to `cp`. I just put it in `pkg/cp` to test things out. The type can also gain more methods as needed, such as a `Getenv` method to replace global `os.Getenv` calls.

Finally, you can look at https://github.com/mvdan/sh/compare/master...93-coreutils to see how I'm hooking it up into my shell interpreter. Pretty straightforward, given that the APIs are fairly similar.

More than happy to send a PR for `cp` and receive reviews if the plan looks reasonable. The only minor trouble is that I'll need to move code from `cmds/core` to `pkg`, so the diffs will be a bit noisy. But hopefully that's relatively easy to review with git's "moved lines" diffs.

mvdan, 2023-01-22:
I forgot to mention one quirk: your module currently depends on `mvdan.cc/sh/v3`, and my tentative plan in my module is to add an `interp/coreutils` package which imports `pkgs/...`, meaning that my module will start depending on yours too.

Module dependency cycles are allowed in Go, but they contain many tricky downsides so I'd like to avoid it if possible. If our modules are cyclic, there will be an ever-growing downward spiral of `require` pointers in MVS, bouncing back and forth between the two projects until the point where the cycle was introduced. If either of our projects requires a dependency at a minimum version (per MVS), then neither of our projects can easily use an _older_ version of that same dependency, because the module graph requires the newer version that's tangled up in that downward spiral. I've had this happen before, and it's rather annoying to untangle.

The question becomes: should I be the one to carve `interp/coreutils` into a separate Go module, or should I propose carving out u-root into two modules?

I admit I've personally never been in a situation where my project is both upstream and downstream when compared to another project, so I'm not sure how to proceed :)

mvdan, 2025-12-03:
@andreynering is doing work in this space now, and I'm not - so I don't think this issue is useful anymore.