u-root/u-root #2529: pkg/cpio: improve robustness of parsing records

RiSKeD, 2022-10-17:
I've implemented fuzzing tests for the cpio pkg and encountered a panic when trying to read invalid records. 
Reading an invalid cpio record like the example in  [PR 2528](https://github.com/u-root/u-root/pull/2528) results in a panic.
An index overflows if encountering an invalid name length of zero in the record:
`	info.Name = string(nameBuf[:hdr.NameLength-1])`

You can reproduce the bug by running the following in pkg/cpio on this branch [PR 2528](https://github.com/u-root/u-root/pull/2528):
```
go test 
--- FAIL: FuzzReadWriteNewc (0.00s)
    --- FAIL: FuzzReadWriteNewc/d463dcdf6acb57ea62ec29d8f45d8976ab187f85ab28565ce8ff7bd906a651ed (0.00s)
panic: runtime error: slice bounds out of range [:4294967295] with capacity 0 [recovered]
        panic: runtime error: slice bounds out of range [:4294967295] with capacity 0

goroutine 23 [running]:
testing.tRunner.func1.2({0x5c28e0, 0xc000020528})
        /home/l1x0r/.gvm/gos/go1.19.2/src/testing/testing.go:1396 +0x24e
testing.tRunner.func1()
        /home/l1x0r/.gvm/gos/go1.19.2/src/testing/testing.go:1399 +0x39f
panic({0x5c28e0, 0xc000020528})
        /home/l1x0r/.gvm/gos/go1.19.2/src/runtime/panic.go:884 +0x212
github.com/u-root/u-root/pkg/cpio.(*reader).ReadRecord(0xc0000fe4b0)
        /home/l1x0r/go/src/github.com/RiSKeD/u-root/pkg/cpio/newc.go:308 +0x8e8
github.com/u-root/u-root/pkg/cpio.EOFReader.ReadRecord({{0x616500?, 0xc0000fe4b0?}})
        /home/l1x0r/go/src/github.com/RiSKeD/u-root/pkg/cpio/utils.go:83 +0xa4
github.com/u-root/u-root/pkg/cpio.ForEachRecord({0x6168c0, 0xc00007cdb0}, 0xc0001bf360)
        /home/l1x0r/go/src/github.com/RiSKeD/u-root/pkg/cpio/utils.go:186 +0x84
github.com/u-root/u-root/pkg/cpio.ReadAllRecords({0x6168c0?, 0xc00007cdb0?})
        /home/l1x0r/go/src/github.com/RiSKeD/u-root/pkg/cpio/utils.go:176 +0x53
github.com/u-root/u-root/pkg/cpio.FuzzReadWriteNewc.func1(0xc0000d1ba0, {0xc00002c480, 0x73, 0x80})
        /home/l1x0r/go/src/github.com/RiSKeD/u-root/pkg/cpio/newc_test.go:530 +0xde
reflect.Value.call({0x59f1e0?, 0x5e24c0?, 0x13?}, {0x5d2dc4, 0x4}, {0xc0000fe450, 0x2, 0x2?})
        /home/l1x0r/.gvm/gos/go1.19.2/src/reflect/value.go:584 +0x8c5
reflect.Value.Call({0x59f1e0?, 0x5e24c0?, 0x51b?}, {0xc0000fe450?, 0x6e4a78?, 0x708ca0?})
        /home/l1x0r/.gvm/gos/go1.19.2/src/reflect/value.go:368 +0xbc
testing.(*F).Fuzz.func1.1(0xc0000e45a0?)
        /home/l1x0r/.gvm/gos/go1.19.2/src/testing/fuzz.go:337 +0x231
testing.tRunner(0xc0000d1ba0, 0xc0000e8630)
        /home/l1x0r/.gvm/gos/go1.19.2/src/testing/testing.go:1446 +0x10b
created by testing.(*F).Fuzz.func1
        /home/l1x0r/.gvm/gos/go1.19.2/src/testing/fuzz.go:324 +0x5b9
exit status 2
```

Any record with a name length of zero is invalid and should be rejected (at least a trailing NUL should be expected: see [CPIO format](https://www.systutorials.com/docs/linux/man/5-cpio/)). I've fixed the bug locally but quickly ran into new errors. The reading of cpio records needs to be made more robust in general and fuzzed again afterward for a sufficient time until no more errors occur.

RiSKeD, 2022-10-26:
Fixed the some of the bugs with #2528. More works needs to be done though...

fotonick, 2023-07-07:
> More work needs to be done though...

@RiSKeD, in #2528, it sounds like you can no longer find any bugs after a good, long fuzzing run. Can this issue be closed?

RiSKeD, 2023-07-10:
Sure, this issue can be closed as no other bugs have been found for a while.