u-root/u-root #2590: Strace package doesn't work with Chrome/Firefox

robertmin1, 2022-12-30:
When using the Strace package to trace Chromium-browser or Firefox, it fails and produces the error: "Trace/breakpoint trap (core dumped)"
Additionally using the program in [u-root](https://github.com/u-root/u-root)/[cmds](https://github.com/u-root/u-root/tree/main/cmds)/[core](https://github.com/u-root/u-root/tree/main/cmds/core)/strace/ still produces the same error. When you monitor the system processes, you can notice the program runs for a few seconds, then crashes. The application (Chromium browser or Firefox) doesn't open

Trace Chromium-browser or Firefox using the U-root strace package. 

The tracing to happen without crashing. The application (Chromium browser or Firefox) to open, similar to the C-based strace.

Other smaller programs that I tested with the u-root package also worked without any problems.

hugelgupf, 2023-01-02:
Heh. Stracing chromium or Firefox is certainly more hardcore than I tested during development. I imagine they make use of every feature under the sun, definitely including ptrace itself.

JeremyRand, 2023-01-02:
It might be interesting to grep the source code, bug tracker, and/or Git history of the C-based strace to see if any mention of Firefox or Chromium related workarounds (or more generally workarounds related to sandboxes) shows up.

Also I wonder if the behavior is different when Chromium is run with the sandbox disabled, or Firefox is run with e10s disabled.

@robertmin1 Any chance you could edit your OP to include the full output of both Firefox and Chromium under u-root's strace?  I wonder if there's a hint earlier in the log, or whether the Firefox and Chromium logs are subtly different.

robertmin1, 2023-01-04:
Added the log for both programs

JeremyRand, 2023-03-19:
@robertmin1 Can you confirm whether the issue still happens when Chromium is run with the `--no-sandbox` argument?

robertmin1, 2023-05-21:
Fixed by #2671

closed 2023-05-21 by robertmin1