u-root/u-root #2591: Make gosh the default shell

jensdrenhaus, 2023-01-05:
Since gosh has been polished and quirks like tab completion are fixed now, should it become the default shell in builds instead of elvish?

hugelgupf, 2023-01-05:
Can you remind me why we previously had to backtrack? Link to that discussion?

fotonick, 2023-01-06:
I've had questions from some u-root users (not necessarily blocking): What is the status of Ctrl-C handling and command history in gosh?

MDr164, 2023-01-09:
As of right now Ctrl+C will be sent to the gosh process and exist it which is not ideal. We can use code like this to handle it:
```
c := make(chan os.Signal, 1)
signal.Notify(c, os.Interrupt)
go func(){
    for sig := range c {
        // sig is a ^C, handle it
    }
}()
```
The question would be: How do we want to handle it? Simply clear the prompt? Open a new prompt? Ask the user whether ctrl+c should cause the shell to exit?

And for command history: I got something in the works for that, can open a PR this week.

fotonick, 2023-01-10:
Thanks, Marvin. That these features are close is happy news.

Probably the most important Ctrl+c behavior is that while a subprocess is running should pass the interrupt to the subprocess. If no subprocess, then it doesn't matter much. zsh seems to treat Ctrl+c similar to hitting return on an empty line, ignoring whatever text might have been entered already.

History definitely wouldn't be a showstopper, but I'm thrilled that you have something in the works. Looking forward to seeing it.

Changing the default shell seems very user-visible. Worth highlighting prominently in a u-root release along with the impacts (features, image size).

mvdan, 2023-03-22:
:wave: happy to help where I can as the upstream for the shell parser and interpreter. PRs are always welcome, if there are any changes that need to happen in either of those.

I think u-root maintaining its own interactive shell on top of the parser and interpreter is the right approach, as it allows for quick iteration and making design decisions like how to implement readline support. Once it gets polished and becomes the default, I could remove my [original and super-minimal cmd/gosh](https://github.com/mvdan/sh/blob/master/cmd/gosh/main.go), since it's currently only good for quick tests or debugging.

Or, if you would be up for it, I'd be happy to have your gosh upstreamed to replace mine :) For example, see https://github.com/u-root/u-root/issues/2527, where it seems like the two projects might otherwise become upstreams and downstreams of each other at the same time.

MDr164, 2023-03-22:
There is currently an open PR at #2606 which deals with the quirks currently present. I used bubbline there which is aiming at being a feature complete readline replacement in go. You can have a look there and let me know if there are better approaches that come to your mind.

fotonick, 2023-07-07:
Gosh made default in #2598. Closing.

closed 2023-07-07 by fotonick