u-root/u-root #2667: Gosh is hard to build tests for

rminnich, 2023-04-30:
Gosh is so smart it is stupid. It's just about impossible to write scripting tests for it because it's so busy
trying to autocomplete. Further, if you do a simple thing like this:
cat | gosh
gosh won't run the commands until cat exits. 
cat | ./gosh
date
ls
echo this sucks
^D
Sun Apr 30 07:43:26 PDT 2023
completer.go            gosh                    gosh_test.go
completer_test.go       gosh.go                 testdata
this sucks

bash does not have this problem:
cat | bash
date
Sun Apr 30 07:42:44 PDT 2023
^D

elvish, similarly, works correctly with cat | elvish.

**Describe the solution you'd like**
gosh should be not so smart.

This problem is causing real trouble for test writers.

jelischer, 2023-05-01:
just having some local variable you cna set to turn it off would be perfect.
There seems ot be another Gosh on the internet and it turns off autocompletion with 
export GOSH_NO_TAB_COMPLETION=1
or so google says.  Something similar would be nice.  
The use case is running automated test scripts (e.g. with expect)

jelischer, 2023-05-01:
also I thought setting the TERM type to 'dumb' might do something but as far as I could see that did nothing to change it.

10000TB, 2023-05-01:
it would also help provide an option to turn it off, e.g. https://github.com/u-root/u-root/blob/main/cmds/core/gosh/gosh.go#L115

MDr164, 2023-05-02:
I'll add a PR to turn off the tab completer during runtime as that's a quick and small change. Btw would adding a compile time switch to not compile the completer in a temporary fix for your dependency evaluation @10000TB? That way the new bubbline library code will not even be inside the binary in the end if so desired.

10000TB, 2023-05-03:
Hi Marvin, no sadly. As long as it is part of 'import' list, it is a dependency

fotonick, 2023-07-07:
I think this was fixed by #2684, in which Ron added a `-interactive` flag. Reopen if I'm mistaken.

closed 2023-07-07 by fotonick