u-root/u-root #2669: wget -O needs to be passed as first argument

tlaurion, 2023-05-01:
https://github.com/DeterminateSystems/nix-installer/blob/5a58cf20bec1b34f62e11709a60ed60d2d640353/nix-installer.sh#L314 doesn't work.

Only `wget -O /tmp/nix-installer https://install.determinate.systems/nix/tag/v0.8.0/nix-installer-x86_64-linux` works, but most tools pass -O after URL, not before.

`-O` should not be positional and only working if provided as first argument...

---
When applied on top of #2668 manully packed under u-root cpio:
```
~ # wget  https://install.determinate.systems/nix/tag/v0.8.0/nix-installer-x86_64-linux -O -O /tmp/nix-installer
~ # ls /tmp/nix-installer 
ls: /tmp/nix-installer: No such file or directory
```

rminnich, 2023-05-01:
I see -O twice, are you sure it did not pull it down to a file name -O?

rminnich, 2023-05-01:
it's not that it is positional, it's that the go flag package takes the standard of switches first and then args. 

wget's rules around args are very non-standard, but we have little choice but to honor them, including the -- I suppose. Quite the mess.

rminnich, 2023-05-01:
man wget | grep '^  *--'|wc
wow. 174 options. 
This is truly a work of art.

tlaurion, 2023-05-01:
@rminnich yeah. Idea i not to try to replicate all of them, but i'm not sure it's acceptable that -O is not respected. Nothing can be done here?

tlaurion, 2023-05-02:
Fixed per https://github.com/u-root/u-root/commit/83a242bbeb93bf42b0d9aa400bff33532d32f59b

closed 2023-05-02 by tlaurion