u-root/u-root #2687: Squashfs output support and stripping of files added content

tlaurion, 2023-05-27:
**Is your feature request related to a problem? Please describe.**

As of today, only initrd output is supported.
One of the interesting feature of u-root is tonpack additional files and directories and their dependencies (ld) 

**Describe the solution you'd like**
One extension feature of u-root could be to pack u-root content as sqashfs content. Ideally, also by stripping binaries and libraries of their symbols prior or creating the squashfs

That way, a linuxboot enabled system could actually load additionally detached signed (or not) squasfsh file and apply it on top of initramfs prior of launching init.

**Describe alternatives you've considered**
Tinycore from exp/tcz, but as of now not really actually working since post-install steps are not applied.

**Additional context**
u-root as of today is nondeterministic.
Even if using one commitnof u-root, dependencies could change. While this is worked on, nix as an host to build u-root could be used to pack host's stripped binaries and libraries to create needed Recovery shell.

hugelgupf, 2023-05-28:
> u-root as of today is nondeterministic.
> Even if using one commit of u-root, dependencies could change.

What do you mean by this? u-root's dependencies are pinned, and our [tests](https://github.com/u-root/u-root/blob/deb77c795abbc4d22a320e6766301d0ef329abfc/uroot_test.go#L361) test that building the same initramfs twice at the same commit on the same machine comes out bit for bit reproducibly the same. Have we overlooked some cases?

tlaurion, 2023-05-29:
@hugelgupf I do see any pinning for random example https://github.com/u-root/u-root/blob/main/pkg/boot/kexec/kexec_load_linux.go#L12

I get that at a moment in time, if one builds the same initramfs twice it will pass the test. My point, irrelevant to subject of this ticket is if someone wants to rebuild same commit in 6 months it won't produce the same initramfs because dependencies might evolve as well since not pinned to dependencies commit, simply downloading latest available version.

hugelgupf, 2023-05-29:
> @hugelgupf I do see any pinning for random example https://github.com/u-root/u-root/blob/main/pkg/boot/kexec/kexec_load_linux.go#L12
> 
> I get that at a moment in time, if one builds the same initramfs twice it will pass the test. My point, irrelevant to subject of this ticket is if someone wants to rebuild same commit in 6 months it won't produce the same initramfs because dependencies might evolve as well since not pinned to dependencies commit, simply downloading latest available version at moment of building.

I see what you mean -- though that's a Go problem with Minimum Version Selection in Go modules. I hadn't realized that Go lost that with the switch from vendoring to modules. (We still have the vendor directory, so theoretically u-root hasn't lost it if you build with modules disabled.)

tlaurion, 2023-12-28:
> I see what you mean -- though that's a Go problem with Minimal Version Selection in Go modules. I hadn't realized that Go lost that with the switch from vendoring to modules. (We still have the vendor directory, so theoretically u-root hasn't lost it if you build with modules disabled.)

@hugelgupf how would one would do that?

> squashfs

Initramfs are ok, squashfs images are better (overlay fs, can expend and overwrite ramfs/tmpfs content)

orangecms, 2025-11-12:
Regarding the question of disabling modules:
```sh
GO111MODULE=off go ...
```

Not too intuitive. 😅

https://go.dev/ref/mod#mod-commands 👈🧐