u-root/u-root #2763: UKI support

orangecms, 2023-09-14:
Patches pending to add UKI support in the kernel:
https://lore.kernel.org/lkml/20230911052535.335770-1-kernel@jfarr.cc/

Discussion is here:
https://github.com/systemd/systemd/issues/28538

When distros start using UKI, kexec userspace will need some adjustments to support it.
For details, see the kernel patches and the discussion.

orangecms, 2023-09-14:
For what UKI is, see also https://media.ccc.de/v/all-systems-go-2023-185-unified-kernel-images-ukis-

dpeckett, 2023-10-23:
For reference this already works today with the existing kexec_file_load() boot implementation. Supporting this for the zerocopy kexec() implementation will be much more complicated.

```
CONFIG_KEXEC_FILE=y
CONFIG_KEXEC_SIG=y
CONFIG_KEXEC_SIG_FORCE=y
CONFIG_KEXEC_BZIMAGE_VERIFY_SIG=y
CONFIG_SIGNED_PE_FILE_VERIFICATION=y
```

leongross, 2024-10-02:
I would like to pick this issue up again. I heard that the Nix Folks like the Nix-UKI-systemd eco system since it integrates seamlessly. 
Now that Nix support for u-root is coming up we might consider working on this integration to be part of that ecosystem. As of now, I do not know a lot about the constraints u-root has to adapt to, to be a viable option for the UKI initrd. Maybe someone more knowledgeable like @orangecms could outline a small roadmap to do so?