u-root/u-root #2785: Ci-Fuzz Build failing

RiSKeD, 2023-10-19:
Currently, the CI-Fuzz build has been failing due to 3 fuzzing targets. 
The build should be fixed by https://github.com/google/oss-fuzz/pull/11139 (excluding the broken targets).
I will have a deeper look into the reasons they are failing later on.

hugelgupf, 2024-03-21:
I'm going to make fuzzing fail again with #2951. We need a different solution that actually works with Go fuzzing

RiSKeD, 2024-03-21:
Check out https://github.com/google/oss-fuzz/issues/7923#issuecomment-1206672834 to see why some fuzzing targets are failing. For more details on necessary changes, visit https://github.com/AdamKorcz/go-118-fuzz-build?tab=readme-ov-file#to-dos.
Until these issues are fixed, we cannot guarantee stable builds for our fuzzing targets...