u-root/u-root #2852: Race condition in `sshd`.

Foxboron, 2024-01-13:
**Describe the bug**
When using `x/crypto/ssh` to connect with `sshd` there is no output from the session with `CombinedOutput`. And manually using OpenSSH to connect to `sshd` *sometimes* doesn't give any output back.

I suspect there is a race condition somewhere but I have not managed to figure out where.

**To Reproduce**
Start `sshd` and use `x/crypto/ssh` to open a session and use `CombinedOutput`. It will fail with `EOF`.

**Expected behavior**
The command should return something.

**Additional context**
N/A

Foxboron, 2024-01-13:
My running theory is that using `e.Process.Wait` instead of `e.Wait()` is the main issue, as we are exiting the before everything is written to stdout/stdin/stderr.

I'm at a loss on how we should properly fix this, but I'm wondering if just execing with `CombinedOutput` and then `io.Copy` it to `c` would be a better solution? Currently it seems `e.Wait()` chokes on the channel we are giving it.

Foxboron, 2024-01-13:
cc @bluecmd  :)

binjip978, 2024-04-07:
I had similar issue with EOF when I was trying to write test for this command: https://github.com/u-root/u-root/pull/2963 and initially I was trying combinedOutput or Output and got only EOF. I tested sshd command with other ssh clients and saw no issues, but what seemed to work is to pipe Stdout and then I got an output. But still I have linux box where this test fails reliably :)

closed 2024-12-26 by binjip978