u-root/u-root #2938: Error during uinit, but still able to boot

HashFail, 2024-02-29:
**Describe the bug**
I'm experiencing a fatal error when u-root starts up. I am building u-root with the following command:

```
u-root -build=bb -files "startup.sh:etc/startup.sh" -uinitcmd="/etc/startup.sh" \
		github.com/u-root/u-root/cmds/core/* \
		github.com/u-root/u-root/cmds/boot/*
```

The contents of `startup.sh` are:

```
#!/bin/sh
echo "Booting...\n"
sleep 10
boot
echo "Boot failed."
```

The initramfs is then built into a UEFI image (I am running u-root as a LinuxBoot bootloader).

When I flash the BIOS and I get the following output (after many messages from the kernel):

```
2024/02/28 22:46:16 Welcome to u-root!
                              _
   _   _      _ __ ___   ___ | |_
  | | | |____| '__/ _ \ / _ \| __|
  | |_| |____| | | (_) | (_) | |_
   \__,_|    |_|  \___/ \___/ \__|

sd 1:0:0:0: [sda] Optimal transfer size 33553920 bytes not a multiple of physical block size (4096 bytes)
sd 1:0:0:0: [sda] Optimal transfer size 33553920 bytes not a multiple of physical block size (4096 bytes)

runtime: failed to create new OS thread (have 3 already; errno=11)
runtime: may need to increase max user processes (ulimit -u)
fatal error: newosproc

runtime stack:
runtime.throw({0xccf296?, 0xc000058400?})
        runtime/panic.go:1077 +0x5c fp=0x7fffc09b3e78 sp=0x7fffc09b3e48 pc=0x44149c
runtime.newosproc(0xc000058400)
        runtime/os_linux.go:195 +0x125 fp=0x7fffc09b3ed0 sp=0x7fffc09b3e78 pc=0x43d8a5
runtime.newm1(0xc000058400)
        runtime/proc.go:2455 +0xcf fp=0x7fffc09b3f10 sp=0x7fffc09b3ed0 pc=0x447b2f
runtime.newm(0xe2e0297f4?, 0x7fffc09b3f78?, 0x4110f3?)
        runtime/proc.go:2430 +0x113 fp=0x7fffc09b3f58 sp=0x7fffc09b3f10 pc=0x447a13
runtime.startm(0xc000042500, 0x1, 0x0)
        runtime/proc.go:2624 +0x16f fp=0x7fffc09b3fa0 sp=0x7fffc09b3f58 pc=0x447f8f
runtime.wakep()
        runtime/proc.go:2760 +0x87 fp=0x7fffc09b3fd0 sp=0x7fffc09b3fa0 pc=0x4483a7
runtime.newproc.func1()
        runtime/proc.go:4487 +0x54 fp=0x7fffc09b4000 sp=0x7fffc09b3fd0 pc=0x44bbb4
traceback: unexpected SPWRITE function runtime.systemstack
runtime.systemstack()
        runtime/asm_amd64.s:509 +0x4a fp=0x7fffc09b4010 sp=0x7fffc09b4000 pc=0x474a2a

goroutine 1 [running, locked to thread]:
runtime.systemstack_switch()
        runtime/asm_amd64.s:474 +0x8 fp=0xc0000525c8 sp=0xc0000525b8 pc=0x4749c8
runtime.newproc(0x18000000000?)
        runtime/proc.go:4480 +0x4b fp=0xc000052600 sp=0xc0000525c8 pc=0x44bb2b
runtime.init.6()
        runtime/proc.go:310 +0x1a fp=0xc000052618 sp=0xc000052600 pc=0x443dfa
runtime.doInit1(0x132f2e0)
        runtime/proc.go:6740 +0xda fp=0xc000052748 sp=0xc000052618 pc=0x450f5a
runtime.doInit({0x132cee0, 0x2, 0x0?})
        runtime/proc.go:6707 +0x2d fp=0xc000052768 sp=0xc000052748 pc=0x450e2d
runtime.main()
        runtime/proc.go:198 +0xe5 fp=0xc0000527e0 sp=0xc000052768 pc=0x443ac5
runtime.goexit()
        runtime/asm_amd64.s:1650 +0x1 fp=0xc0000527e8 sp=0xc0000527e0 pc=0x476821

goroutine 2 [runnable]:
runtime.forcegchelper()
        runtime/proc.go:313 fp=0xc000052fe0 sp=0xc000052fd8 pc=0x443e20
runtime.goexit()
        runtime/asm_amd64.s:1650 +0x1 fp=0xc000052fe8 sp=0xc000052fe0 pc=0x476821
created by runtime.init.6 in goroutine 1
        runtime/proc.go:310 +0x1a
```

Sometimes, the error I see is instead

```
2024/02/28 18:41:57 fork/exec /bin/uinit: resource temporarily unavailable
```

u-root drops to a shell after this, and I can boot successfully using `boot`

If I do not supply the `uinitcmd` argument, or change the value to `/bbin/boot` to attempt to jump right into `boot`, I instead see a message asking me if I would like to reboot or enter the shell. However, it seems to default to the shell without waiting for my response. Once in the shell, I can again run `boot` and the ubuntu partition on my SSD boots just fine.

How can I resolve this so that I can boot directly into my OS without user interaction? In case the kernel messages are helpful, here is the full output of one such boot: https://pastebin.com/L4hjf2eC

hugelgupf, 2024-02-29:
That’s a cool bug. 

Can you add -v (just -v) to the kernel cmdline args? It’ll make the init process not quiet the kernel prints once userland is started and we can see if there’s something interesting happening during that crash.

HashFail, 2024-02-29:
@hugelgupf thanks for the response! I now see the following additional output:

```
2024/02/29 20:35:28 Welcome to u-root!
                              _
   _   _      _ __ ___   ___ | |_
  | | | |____| '__/ _ \ / _ \| __|
  | |_| |____| | | (_) | (_) | |_
   \__,_|    |_|  \___/ \___/ \__|

u-root init: error creating mount -t "securityfs" -o  "securityfs" "/sys/kernel/security" flags 0x0: no such file or directory
u-root init: error creating mount -t "securityfs" -o  "securityfs" "/sys/kernel/security" flags 0x0: no such file or directory
u-root init [optional]: warning creating mount -t "tmpfs" -o  "cgroup" "/sys/fs/cgroup" flags 0x0: no such file or directory
u-root init [optional]: warning creating mount -t "tmpfs" -o  "cgroup" "/sys/fs/cgroup" flags 0x0: no such file or directory
usb 2-3: new high-speed USB device number 2 using xhci_hcd
usb 2-3: new high-speed USB device number 2 using xhci_hcd
u-root init [optional]: warning creating dir "/sys/fs/cgroup/memory" (mode 0555): mkdir /sys/fs/cgroup: operation not permitted
u-root init [optional]: warning creating dir "/sys/fs/cgroup/memory" (mode 0555): mkdir /sys/fs/cgroup: operation not permitted
u-root init [optional]: warning creating dir "/sys/fs/cgroup/freezer" (mode 0555): mkdir /sys/fs/cgroup: operation not permitted
u-root init [optional]: warning creating dir "/sys/fs/cgroup/freezer" (mode 0555): mkdir /sys/fs/cgroup: operation not permitted
u-root init [optional]: warning creating dir "/sys/fs/cgroup/devices" (mode 0555): mkdir /sys/fs/cgroup: operation not permitted
u-root init [optional]: warning creating dir "/sys/fs/cgroup/devices" (mode 0555): mkdir /sys/fs/cgroup: operation not permitted
u-root init [optional]: warning creating dir "/sys/fs/cgroup/cpu,cpuacct" (mode 0555): mkdir /sys/fs/cgroup: operation not permitted
u-root init [optional]: warning creating dir "/sys/fs/cgroup/cpu,cpuacct" (mode 0555): mkdir /sys/fs/cgroup: operation not permitted
u-root init [optional]: warning creating dir "/sys/fs/cgroup/blkio" (mode 0555): mkdir /sys/fs/cgroup: operation not permitted
u-root init [optional]: warning creating dir "/sys/fs/cgroup/blkio" (mode 0555): mkdir /sys/fs/cgroup: operation not permitted
u-root init [optional]: warning creating dir "/sys/fs/cgroup/cpuset" (mode 0555): mkdir /sys/fs/cgroup: operation not permitted
u-root init [optional]: warning creating dir "/sys/fs/cgroup/cpuset" (mode 0555): mkdir /sys/fs/cgroup: operation not permitted
u-root init [optional]: warning creating dir "/sys/fs/cgroup/pids" (mode 0555): mkdir /sys/fs/cgroup: operation not permitted
u-root init [optional]: warning creating dir "/sys/fs/cgroup/pids" (mode 0555): mkdir /sys/fs/cgroup: operation not permitted
usb 2-3: New USB device found, idVendor=0424, idProduct=2660, bcdDevice= 8.01
usb 2-3: New USB device found, idVendor=0424, idProduct=2660, bcdDevice= 8.01
u-root init [optional]: warning creating dir "/sys/fs/cgroup/net_cls,net_prio" (mode 0555): mkdir /sys/fs/cgroup: operation not permitted
u-root init [optional]: warning creating dir "/sys/fs/cgroup/net_cls,net_prio" (mode 0555): mkdir /sys/fs/cgroup: operation not permitted
usb 2-3: New USB device strings: Mfr=0, Product=0, SerialNumber=0
usb 2-3: New USB device strings: Mfr=0, Product=0, SerialNumber=0
init: 2024/02/29hub 2-3:1.0: USB hub found
hub 2-3:1.0: USB hub found
hub 2-3:1.0: 2 ports detected
hub 2-3:1.0: 2 ports detected
 20:35:28 stat /inito: no such file or directory
init: 2024/02/29 20:35:28 stat /bbin/uinit: no such file or directory
init: 2024/02/29 20:35:28 Trying to run /bin/uinit
usb 3-4: new SuperSpeed USB device number 2 using xhci_hcd
usb 3-4: new SuperSpeed USB device number 2 using xhci_hcd
usb 3-4: New USB device found, idVendor=0bda, idProduct=0329, bcdDevice=29.08
usb 3-4: New USB device found, idVendor=0bda, idProduct=0329, bcdDevice=29.08
usb 3-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
usb 3-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
usb 3-4: Product: USB3.0-CRW
usb 3-4: Product: USB3.0-CRW
usb 3-4: Manufacturer: Generic
usb 3-4: Manufacturer: Generic
usb 3-4: SerialNumber: 29203008282014000
usb 3-4: SerialNumber: 29203008282014000
usb-storage 3-4:1.0: USB Mass Storage device detected
usb-storage 3-4:1.0: USB Mass Storage device detected
scsi host0: usb-storage 3-4:1.0
scsi host0: usb-storage 3-4:1.0
usb 2-6: new low-speed USB device number 3 using xhci_hcd
usb 2-6: new low-speed USB device number 3 using xhci_hcd
usb 2-6: New USB device found, idVendor=1c4f, idProduct=0016, bcdDevice= 1.10
usb 2-6: New USB device found, idVendor=1c4f, idProduct=0016, bcdDevice= 1.10
usb 2-6: New USB device strings: Mfr=1, Product=2, SerialNumber=0
usb 2-6: New USB device strings: Mfr=1, Product=2, SerialNumber=0
usb 2-6: Product: USB Keyboard
usb 2-6: Product: USB Keyboard
usb 2-6: Manufacturer: SIGMACHIP
usb 2-6: Manufacturer: SIGMACHIP
input: SIGMACHIP USB Keyboard as /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/0003:1C4F:0016.0001/input/input0
[... 47 lines cut ...]
usb 2-13: Manufacturer: Generic
usb 2-13: Manufacturer: Generic
input: Generic USB Keyboard as /devices/pci0000:00/0000:00:14.0/usb2/2-13/2-13:1.0/0003:040B:2000.0003/input/input3
input: Generic USB Keyboard as /devices/pci0000:00/0000:00:14.0/usb2/2-13/2-13:1.0/0003:040B:2000.0003/input/input3
hid-generic 0003:040B:2000.0003: input,hidraw2: USB HID v1.10 Keyboard [Generic USB Keyboard] on usb-0000:00:14.0-13/input0
[... cut ...]
scsi 0:0:0:0: Direct-Access     Generic- SD/MMC CRW       1.00 PQ: 0 ANSI: 6
sd 0:0:0:0: [sdb] Media removed, stopped polling
sd 0:0:0:0: [sdb] Media removed, stopped polling
sd 0:0:0:0: [sdb] Attached SCSI removable disk
sd 0:0:0:0: [sdb] Attached SCSI removable disk
2024/02/29 20:36:24 fork/exec /bin/uinit: resource temporarily unavailable
init: 2024/02/29 20:36:31 Shell exited, exit status 1
init: 2024/02/29 20:36:31 stat /buildbin/uinit: no such file or directory
init: 2024/02/29 20:36:31 Trying to run /bin/defaultsh
```

hugelgupf, 2024-02-29:
It kinda feels like a fork-bomb. I'll try to repro in a VM test.

hugelgupf, 2024-03-02:
Definitely a fork bomb, I have a repro https://github.com/hugelgupf/u-root/blob/forkbomb/integration/generic-tests/fork_bomb_test.go (no need to fork bomb your own system when it can be done in a VM).

Looks like it's an issue with how gosh deals with hash-bang invocations, because when I use the equivalent of `-uinitcmd="gosh script.sh"` it works just fine (which is the workaround you can use for now)

HashFail, 2024-03-04:
Thanks a lot friend! That did the trick.

hugelgupf, 2024-03-04:
I got a draft to fix the hash bang issue but still working on better testing. Keeping this open till I’m able to do that.

HashFail, 2024-03-04:
Will do

PR #2942 (2024-03-04): Fix fork bomb
And a couple of other ways that gosh arg interpretation was wrong.

Fixes #2938.
closed 2024-03-04 by hugelgupf