u-root/u-root #2974: ping localhost doesn't work in a vfkit vm?

rjkroege, 2024-04-26:
**Describe the bug**
I create a Linux vm using `vfkit` on an Apple Silicon Mac. I booted such a vm using the test `puipui` kernel from [Code-Hex/vz: Create virtual machines and run Linux-based operating systems in Go using Apple Virtualization.framework.](https://github.com/Code-Hex/vz) and an `initrd` build with `u-root`.  

**To Reproduce**
Boot the vm. `ping 127.0.0.1`. Note the following unexpected output

```
% ping 127.0.0.1
64 bytes from 127.0.0.1: icmp_seq=1 time=75.875µs
1970/01/01 00:01:38 ping failed: got seq 1; want 2
```

**Expected behavior**
I expect `ping 127.0.0.1` to indicate that it's connected. For example, if I boot the `puipui` kernel (same kernel as above) with its associated `initrd`, 

```
~ # ping 127.0.0.1
127.0.0.1 is alive!
```

So: `ping` from busy box works as expected while `u-root` ping does not where kernel, vm and vm launch configuration are held constant.

Conclusion: there is something not-right happening with my `u-root` to setup the network. Any helpful comments would be appreciated. Otherwise, I will continue to explore this and update the bug.

binjip978, 2024-04-26:
Yeah, this is a problem with u-root ping, this can be reproduced on any machine, you don't need a VM. To make it work on localhost is more tricky. It will work with -c=1 but no more.

rjkroege, 2024-04-27:
Ah. Thanks for the update.

I should probably note in passing that I discovered this bug while trying to make the network work in my vm and stopped at "if I can't even get `ping` to localhost to work, then there's no hope for more general networking"