u-root/u-root #2979: tinygo support

leongross, 2024-04-30:
**Is your feature request related to a problem? Please describe.**
Currently, many of the busybox commands fail to build using tinygo.
This issue should cover the current process of enabling all subcommands to be built using tinygo.

**Describe the solution you'd like**
While enabling more and more commands the list of working commands in this issue will be updated. Maybe it is also a good idea to track the PRs and issues for tinygo that cover current / past changes.

**Additional context**
# building
Since atm the tinygo support is lacking, many commands got the `!tinygo` build tag.
This prohibits building, so anyone recreating the build steps should be aware that this has to be removed for each command.

Further, this `u-root` features a tool named `tinygobb` which should build all the subcommands using tinygo. I had some issues with that so I built a python script that builds each command for a subfolder (here run separately in `core` and `exp`). For completeness' sake, I will add it [here](https://gist.github.com/leongross/3ae8517cfae68d1795ad7fe243efb701) as well.
It also outputs build errors in csv so it can be imported to shared spreadsheet software.

# issues
A list of the more prevalent issues that occur during the build process that are either wip or haven't been handled yet (not exhaustive). They will be updated over time. For a more in-depth analysis see [here](https://github.com/orgs/u-root/projects/2).

* > ld.lld: error: undefined symbol: golang.org/x/sys/unix.Syscall6
* > ld.lld: error: undefined symbol: golang.org/x/sys/unix.Syscall
* > ld.lld: error: undefined symbol: golang.org/x/sys/unix.RawSyscallNoError
* > ld.lld: error: undefined symbol: golang.org/x/sys/cpu.cpuid
* > lld: error: undefined symbol: github.com/klauspost/compress
* net.{UnixConn,UnixListener,ListenUnix,DialUnix,UnixAddr,UDPConn,ListenUDP,ResolveUDPAddr}

# command list
## core
- [x] `backoff`
- [x] `base64`
- [x] `basename`
- [x] `blkid`
- [x] `cat`
- [x] `chmod`
- [x] `chroot`
- [x] `cmp`
- [x] `comm`
- [x] `cp`
- [x] `date`
- [x] `dd`
- [x] `dirname`
- [x] `echo`
- [x] `false`
- [x] `find`
- [x] `free`
- [x] `gpt`
- [x] `grep`
- [x] `hexdump`
- [x] `id`
- [x] `io`
- [x] `kill`
- [x] `lddfiles`
- [x] `ls`
- [x] `lsdrivers`
- [x] `lsmod`
- [x] `man`
- [x] `md5sum`
- [x] `mkdir`
- [x] `mktemp`
- [x] `more`
- [x] `mv`
- [x] `pci`
- [x] `printenv`
- [x] `ps`
- [x] `pwd`
- [x] `readlink`
- [x] `rm`
- [x] `rsdp`
- [x] `scp`
- [x] `seq`
- [x] `shasum`
- [x] `sleep`
- [x] `sort`
- [x] `strings`
- [x] `sync`
- [x] `tail`
- [x] `tar`
- [x] `time`
- [x] `timeout`
- [x] `touch`
- [x] `tr`
- [x] `true`
- [x] `ts`
- [x] `uniq`
- [x] `unmount`
- [x] `unshare`
- [x] `uptime`
- [x] `wc`
- [x] `xargs`
- [x] `yes`

43 not working

## exp
- [x] `acpicat`
- [x] `acpigrep`
- [x] `ansi`
- [x] `cbmem`
- [x] `crc`
- [x] `dmidecode`
- [x] `dumpebda`
- [x] `ectool`
- [x] `ed`
- [x] `fbsplash`
- [x] `fdtdump`
- [x] `field`
- [x] `fixrsdp`
- [x] `forth`
- [x] `freq`
- [x] `lsfabric`
- [x] `readelf`
- [x] `readpe`
- [x] `rush`
- [x] `smn`
- [x] `tac`
- [x] `watch`
- [x] `zbi`
- [x] `zimage`

leongross, 2024-05-21:
Do you have a proposal where we could place this?

leongross, 2024-05-21:
@10000TB https://github.com/u-root/u-root/pull/2987

leongross, 2024-06-25:
With the newest syscall improvement [1] we have a total of `114` commands building in `cmd/core` and `cmd/exp`

[1]: https://github.com/tinygo-org/tinygo/pull/4310

deadprogram, 2026-10-21:
How did I only discover this tracking issue today? In any case, how can I help out?