u-root/u-root #3031: cpio+reproducible is breaking hardlinks

Itxaka, 2024-06-26:
**Describe the bug**
`MakeReproducible` is removing the Nlink values from records, which breaks any hardlinks

**To Reproduce**
Steps to reproduce the behavior:
You got some hard links in there (try fedora and `/usr/sbin/mkfs.ext{2,3,4}` which link to `mke2fs` for example)
Do a cpio of that dir with the `MakeReproducible` method for the records.
Extract the cpio archive and check the hardlinks, they will all be 0 on size and do nothing.

**Expected behavior**
A clear and concise description of what you expected to happen.
As with gnu cpio, those links should work still after archiving

**Additional context**
Add any other context about the problem here.

Test case to reproduce:

Source main.go:
```
package main

import (
	"fmt"
	"github.com/u-root/u-root/pkg/cpio"
	"os"
	"path/filepath"
	"strings"
)

func main() {
	archiver, err := cpio.Format("newc")
	if err != nil {
		fmt.Print(err)
		os.Exit(1)
	}

	cpioFileName := "/tmp/test.cpio"
	cpioFile, err := os.Create(cpioFileName)
	if err != nil {
		fmt.Print(err)
		os.Exit(1)
	}
	defer cpioFile.Close()

	rw := archiver.Writer(cpioFile)
	cr := cpio.NewRecorder()

	if err = os.Chdir("/tmp/test_cpio"); err != nil {
		fmt.Print(err)
		os.Exit(1)
	}

	// Walk through the source directory and add files to the cpio archive
	err = filepath.Walk(".", func(filePath string, fileInfo os.FileInfo, err error) error {
		if err != nil {
			return err
		}

		if strings.Contains(filePath, "initramfs.cpio") {
			return nil
		}

		rec, err := cr.GetRecord(filePath)
		if err != nil {
			return fmt.Errorf("getting record of %q failed: %w", filePath, err)
		}

		if err := rw.WriteRecord(cpio.MakeReproducible(rec)); err != nil {
			return fmt.Errorf("writing record %q failed: %w", filePath, err)
		}

		return nil
	})
	if err != nil {
		fmt.Print(err)
		os.Exit(1)
	}

	if err := cpio.WriteTrailer(rw); err != nil {
		fmt.Print(err)
		os.Exit(1)
	}
}
```

Dir that is gonna be cpio-ed:
```
$ ls -ltrahi /tmp/test_cpio 
total 2,0M
16782 drwxr-xr-x  2 itxaka itxaka   80 jun 26 15:53 .
16783 -rw-r--r--  2 itxaka itxaka 1,0M jun 26 15:53 test2
16783 -rw-r--r--  2 itxaka itxaka 1,0M jun 26 15:53 test1
```

output file is perfectly reproducible as expected, no matter how many times is recreated, sha256sum is always the same.

But extracting it produces broken hardlinks:

```
$ cpio -i < /tmp/test.cpio
2049 blocks
$ ls -ltrahi                
total 1,0M
    1 drwxrwxrwt 22 root   root    620 jun 26 15:57 ..
16808 -rw-r--r--  1 itxaka itxaka    0 jun 26 15:58 test2
16807 -rw-r--r--  1 itxaka itxaka 1,0M jun 26 15:58 test1
16806 drwxr-xr-x  2 itxaka itxaka   80 jun 26 15:58 .
```

As you can see, test2 is now an empty broken file.

Creating a custom `MakeReproducible` that does not set the `r.NLink = 0` makes the extraction work as expected.

It makes the archives still reproducible no matter how many times is recreated:
```
$ sha256sum /tmp/test.cpio && rm /tmp/test.cpio && go run main.go
108d920cb415029573b72b14e61ae03b31b5d99820d877f23c09134e38f9e43f  /tmp/test.cpio

$ sha256sum /tmp/test.cpio && rm /tmp/test.cpio && go run main.go
108d920cb415029573b72b14e61ae03b31b5d99820d877f23c09134e38f9e43f  /tmp/test.cpio
```

And when extracting the cpio archive, the hardlinks are still valid:
```
$ ls -ltrahi  
total 2,0M
    1 drwxrwxrwt 22 root   root    620 jun 26 16:05 ..
16843 -rw-r--r--  2 itxaka itxaka 1,0M jun 26 16:05 test2
16843 -rw-r--r--  2 itxaka itxaka 1,0M jun 26 16:05 test1
16806 drwxr-xr-x  2 itxaka itxaka   80 jun 26 16:05 .
```

Itxaka, 2024-06-26:
Hey, thanks for the explanation. I did not dig much deeper that checking that it was the MakeReproducible function the one that was breaking the hard links, as it was something that was working for us and we just introduced yesterday and noticed this. Also my first solution was to override it and not remove the value for blink and that fixed automatically so never digged any further!

Let me know if you need more info about this. I also wondered how it worked with the inode gone, how does it know what files are a hardlink of each other as that's the only reference right? Having the same inode number just means that data is hardlinked. But as I said I didn't go any further that that so no idea.

Let me know if you want more feedback or to test something else, this is something we use on Kairos to generate an initrd from a normal rootfs and generate an USI (unified system image) for EFI so we got an ongoing workflow to test the final system and such.

Thanks!

rminnich, 2024-06-26:
I can't repro your bug, can you see if I got this wrong.

```
bash-3.2$ find . -print | ../../cpio o > ../x.cpio
bash-3.2$ cd ../out
bash-3.2$ ../../cpio i < ../x.cpio
bash-3.2$ ls -li ../in ../out
../in:
total 8
23765592 -rw-r--r--  2 rminnich  staff    0 Jun 26 09:45 a
23765592 -rw-r--r--  2 rminnich  staff    0 Jun 26 09:45 b
23765597 -rw-r--r--  1 rminnich  staff    0 Jun 26 09:45 c
23765716 -rw-r--r--  1 rminnich  staff  521 Jun 26 09:48 typescript

../out:
total 8
23765733 -rw-r--r--  2 rminnich  staff   0 Jun 26 09:48 a
23765733 -rw-r--r--  2 rminnich  staff   0 Jun 26 09:48 b
23765734 -rw-r--r--  1 rminnich  staff   0 Jun 26 09:48 c
23765732 -rw-r--r--  1 rminnich  staff  43 Jun 26 09:48 typescript
```

inumbers for a and b are  the same. 

If you run this command, what happens.

rminnich, 2024-06-26:
By the way, the gnu cpio behavior is VERY un-unix-like
```
ronsexcllentmbp:in rminnich$ cpio -ivt < ../x.cpio
drwxr-xr-x   6 rminnich staff           0 Jun 26 09:48 .
-rw-r--r--   1 rminnich staff          43 Jun 26 09:48 typescript
-rw-r--r--   2 rminnich staff           0 Jun 26 09:45 a
-rw-r--r--   1 rminnich staff           0 Jun 26 09:45 c
-rw-r--r--   2 rminnich staff           0 Jun 26 09:45 b link to a
```
See that 'b link to a' -- how do I know it's not a file called 'b link to a'? 
Who writes things like this?

rminnich, 2024-06-26:
OK I can repro your problem. 

What's odd is that whether to create a hard link should be determined by inumber. NLink is not supposed to matter. I'll keep looking. I am now officially puzzled. Thanks for finding this.

rminnich, 2024-06-26:
OK, I'm a bit further. Basically, the way you tell if you have a hard link is you have a zero-length file and an inumber you've processed already.

Just messing around, zero-ing NLink and NOT zeroing Ino seems to be ok. What's funny is GNU CPIO seems to get it wrong. Still looking. back later.

rminnich, 2024-06-26:
can you try this?
```
--- a/pkg/cpio/utils.go
+++ b/pkg/cpio/utils.go
@@ -278,7 +278,10 @@ func Normalize(name string) string {
 // files have unchanged contents, the cpio file it produces will be bit-for-bit
 // identical. This is an essential property for firmware-embedded payloads.
 func MakeReproducible(r Record) Record {
-       r.Ino = 0
+       // Do NOT zero Ino. The Ino is created in a reproducible manner
+       // and a non-zero value is critical for creating hard links when
+       // reading the archive.
+       // r.Ino = 0
        r.Name = Normalize(r.Name)
        r.MTime = 0
        r.UID = 0
```

We need to write a test before we commit any fix however.

rminnich, 2024-06-26:
I've really got no idea why your NLink change worked, it's a mystery.

rminnich, 2024-06-26:
Please see https://github.com/u-root/u-root/pull/3033

Itxaka, 2024-06-26:
Nice catch there! I still wonder why my workaround works... I guess I'll investigate a bit more when I have time to find out why would it 😀

closed 2024-12-26 by binjip978