u-root/u-root #3072: mount -o bind /proc /mnt/proc does not work

CodingVoid, 2024-08-14:
**Describe the bug**
mount -o bind /proc /mnt/proc does not work.
In the coreutils it works fine, but in u-root it does not. It is mainly useful for a chroot.

**To Reproduce**
Steps to reproduce the behavior:
```
mkdir -p /mnt/proc
mount -o bind /proc /mnt/proc
```

**Expected behavior**
It should mount successfully.

**Additional context**
The issue is that if no fstype (via -t) is supplied to u-root we use the TryMount function which in turn fails to recognize the filesystem correctly.
```
mount -t proc -o bind /proc /mnt/proc
```
works fine so the only issue is with detecting the procfs filesystem on the source mountpoint.

CodingVoid, 2024-08-14:
The same also goes for the devtmpfs which is used for /dev

yin-zt, 2025-01-21:
how does u-root recognize the hard disk?