u-root/u-root #3124: data race in netcat

binjip978, 2024-09-29:
**Describe the bug**
A data race in cmds/core/netcat

**To Reproduce**
Steps to reproduce the behavior:
go test -race -run=TestListenForConnections

**Expected behavior**
Test should pass.

**Additional context**
I happened at least once in CI https://app.circleci.com/pipelines/github/u-root/u-root/8062/workflows/77d4702a-ac52-4495-81e0-2af30c4bba1f/jobs/107468?utm_campaign=vcs-integration-link&utm_medium=referral&utm_source=github-checks-link&utm_content=summary

But more or less reliable to reproduce on a slower machine.

```bash
root@epiphyte:~/src/u-root/cmds/core/netcat# go test -race -run=TestListenForConnections
==================
WARNING: DATA RACE
Read at 0x00c000089110 by goroutine 16:
  runtime.mapaccess1_fast32()
      /usr/local/go/src/runtime/map_fast32.go:13 +0x0
  github.com/u-root/u-root/cmds/core/netcat.(*cmd).listenForConnections.func2()
      /root/src/u-root/cmds/core/netcat/listen.go:226 +0x2da
  github.com/u-root/u-root/cmds/core/netcat.(*cmd).listenForConnections.gowrap3()
      /root/src/u-root/cmds/core/netcat/listen.go:237 +0x4c

Previous write at 0x00c000089110 by goroutine 14:
  runtime.mapassign_fast32()
      /usr/local/go/src/runtime/map_fast32.go:93 +0x0
  github.com/u-root/u-root/cmds/core/netcat.(*Connections).Add()
      /root/src/u-root/cmds/core/netcat/listen.go:116 +0xcc
  github.com/u-root/u-root/cmds/core/netcat.(*cmd).listenForConnections()
      /root/src/u-root/cmds/core/netcat/listen.go:192 +0x674
  github.com/u-root/u-root/cmds/core/netcat.TestListenForConnections.func49()
      /root/src/u-root/cmds/core/netcat/listen_test.go:529 +0x464
  testing.tRunner()
      /usr/local/go/src/testing/testing.go:1689 +0x21e
  testing.(*T).Run.gowrap1()
      /usr/local/go/src/testing/testing.go:1742 +0x44

Goroutine 16 (running) created at:
  github.com/u-root/u-root/cmds/core/netcat.(*cmd).listenForConnections()
      /root/src/u-root/cmds/core/netcat/listen.go:195 +0x918
  github.com/u-root/u-root/cmds/core/netcat.TestListenForConnections.func49()
      /root/src/u-root/cmds/core/netcat/listen_test.go:529 +0x464
  testing.tRunner()
      /usr/local/go/src/testing/testing.go:1689 +0x21e
  testing.(*T).Run.gowrap1()
      /usr/local/go/src/testing/testing.go:1742 +0x44

Goroutine 14 (running) created at:
  testing.(*T).Run()
      /usr/local/go/src/testing/testing.go:1742 +0x825
  github.com/u-root/u-root/cmds/core/netcat.TestListenForConnections()
      /root/src/u-root/cmds/core/netcat/listen_test.go:507 +0x27cb
  testing.tRunner()
      /usr/local/go/src/testing/testing.go:1689 +0x21e
  testing.(*T).Run.gowrap1()
      /usr/local/go/src/testing/testing.go:1742 +0x44
==================
==================
WARNING: DATA RACE
Read at 0x00c000089110 by goroutine 18:
  runtime.mapaccess1_fast32()
      /usr/local/go/src/runtime/map_fast32.go:13 +0x0
  github.com/u-root/u-root/cmds/core/netcat.(*cmd).listenForConnections.func2()
      /root/src/u-root/cmds/core/netcat/listen.go:226 +0x2da
  github.com/u-root/u-root/cmds/core/netcat.(*cmd).listenForConnections.gowrap3()
      /root/src/u-root/cmds/core/netcat/listen.go:237 +0x4c

Previous write at 0x00c000089110 by goroutine 16:
  runtime.mapdelete_fast32()
      /usr/local/go/src/runtime/map_fast32.go:273 +0x0
  github.com/u-root/u-root/cmds/core/netcat.(*Connections).Delete()
      /root/src/u-root/cmds/core/netcat/listen.go:123 +0xaa
  github.com/u-root/u-root/cmds/core/netcat.(*cmd).listenForConnections.func2.1()
      /root/src/u-root/cmds/core/netcat/listen.go:199 +0x35
  runtime.deferreturn()
      /usr/local/go/src/runtime/panic.go:602 +0x5d
  github.com/u-root/u-root/cmds/core/netcat.(*cmd).listenForConnections.gowrap3()
      /root/src/u-root/cmds/core/netcat/listen.go:237 +0x4c

Goroutine 18 (running) created at:
  github.com/u-root/u-root/cmds/core/netcat.(*cmd).listenForConnections()
      /root/src/u-root/cmds/core/netcat/listen.go:195 +0x918
  github.com/u-root/u-root/cmds/core/netcat.TestListenForConnections.func49()
      /root/src/u-root/cmds/core/netcat/listen_test.go:529 +0x464
  testing.tRunner()
      /usr/local/go/src/testing/testing.go:1689 +0x21e
  testing.(*T).Run.gowrap1()
      /usr/local/go/src/testing/testing.go:1742 +0x44

Goroutine 16 (finished) created at:
  github.com/u-root/u-root/cmds/core/netcat.(*cmd).listenForConnections()
      /root/src/u-root/cmds/core/netcat/listen.go:195 +0x918
  github.com/u-root/u-root/cmds/core/netcat.TestListenForConnections.func49()
      /root/src/u-root/cmds/core/netcat/listen_test.go:529 +0x464
  testing.tRunner()
      /usr/local/go/src/testing/testing.go:1689 +0x21e
  testing.(*T).Run.gowrap1()
      /usr/local/go/src/testing/testing.go:1742 +0x44
==================
--- FAIL: TestListenForConnections (0.02s)
    --- FAIL: TestListenForConnections/Read_with_data_from_multiple_connections (0.00s)
        testing.go:1398: race detected during execution of test
FAIL
exit status 1
FAIL	github.com/u-root/u-root/cmds/core/netcat	0.053s
```

RiSKeD, 2024-12-02:
Maybe this has been fixed with: https://github.com/u-root/u-root/pull/3183/commits/e3a69edf4f04c573062f8a11c92a2580427c4f91

binjip978, 2024-12-03:
Yeah, looks like fixed.