u-root/u-root #3153: gosh support "cd"

letian0805, 2024-10-29:
[image]

rminnich, 2024-10-30:
go build
Rons-Excellent-Macbook-Pro:gosh rminnich$ ./gosh
> pwd
/Users/rminnich/go/src/github.com/u-root/u-root/cmds/core/gosh
> cd /
> pwd
/
>
> cd /bin/
> pwd
/bin
>

this is on osx

we need more info. What OS?

letian0805, 2024-10-31:
[image]

I am running this in a docker container based on the scratch image.
`/bin/sh` is compiled with the original code, and `/sh` is compiled with my modified code.

binjip978, 2024-10-31:
> ![image](https://private-user-images.githubusercontent.com/14961048/381797574-a01fdcee-3eae-4262-a68d-0d49a681d6ce.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

I you sure that directory exist in your scratch image?

➜  gosh git:(main) ./gosh                            
> cd /hkhkj                                                                                                                                                       
error: exit status 1
> cd /                                                                                                                                                            
> cd /hkhkj                                                                                                                                                       
error: exit status 1
>

rminnich, 2024-10-31:
It would help if you could provide a complete set of commands to reproduce your problem. I think it is a problem in your container. I am unable to reproduce your issue.

rminnich, 2024-10-31:
I am also unclear on what you mean by 'scratch image'. Which one? Can I do a docker pull to try it?

letian0805, 2024-11-01:
> I am also unclear on what you mean by 'scratch image'. Which one? Can I do a docker pull to try it?

The scratch image is provided by Docker naturally, and you can use it directly without pulling it.

```dockerfile
FROM golang:1.22.5-alpine3.20  as build
RUN mkdir /gotmp/ && touch /gotmp/.history
ARG CGO_ENABLED=0
RUN go install -ldflags "-s -w" github.com/u-root/gobusybox/src/cmd/makebb@latest
RUN wget https://codeload.github.com/u-root/u-root/tar.gz/refs/tags/v0.14.0 -O v0.14.0.tar.gz && tar xf v0.14.0.tar.gz
RUN mkdir -p /gobusybox/bin && cd u-root-0.14.0 && makebb "./cmds/core/{gosh,ls}" && cp bb "/gobusybox/" && \
    ln -s /gobusybox/bb /gobusybox/gosh && \
    ln -s /gobusybox/gosh /gobusybox/bin/sh && \
    ln -s /gobusybox/bb /gobusybox/bin/ls

RUN mkdir /gotmp/ && touch /gotmp/.history

FROM scratch

COPY --from=build /gobusybox/ /gobusybox/
COPY --from=build /gobusybox/bin/ /bin/
COPY --from=build /gotmp/ /tmp/
ENV PATH=/bin:/gobusybox
CMD ["/bin/sh"]
```

This is my Dockerfile, you can try it:
```bash
docker build -t gobusybox
docker run -it gobusybox
```
[image]

binjip978, 2024-11-01:
gosh already support builtins:

```go
func isBuiltin(name string) bool {
	switch name {
	case "true", ":", "false", "exit", "set", "shift", "unset",
		"echo", "printf", "break", "continue", "pwd", "cd",
		"wait", "builtin", "trap", "type", "source", ".", "command",
		"dirs", "pushd", "popd", "umask", "alias", "unalias",
		"fg", "bg", "getopts", "eval", "test", "[", "exec",
		"return", "read", "mapfile", "readarray", "shopt":
		return true
	}
	return false
}
```

I think issue happens because of permissions(see my comment in corresponding PR), so it maybe an issue sh but not in u-root.

closed 2024-11-01 by binjip978

letian0805, 2024-11-01:
> gosh already support builtins:
>
> ```go
> func isBuiltin(name string) bool {
> 	switch name {
> 	case "true", ":", "false", "exit", "set", "shift", "unset",
> 		"echo", "printf", "break", "continue", "pwd", [...]

[image]

You are right, I modified the source code and added some logs and found that it was because the USER environment variable was not set. After setting the environment variable, it was normal.

Thanks a lot!