u-root/u-root #3155: tail -f  buf when read last 10 lines

letian0805, 2024-10-30:
[image]

tail -f reads the last 10 lines of the file and outputs the wrong content.

rminnich, 2024-10-30:
```
cat /etc/hosts | tail -f

# serial over lan to rome
10.0.1.128 sol

192.168.42.1 milkv mv

192.168.86.219 relays
192.168.86.250 psz pistarzero

192.168.0.240 i11 u0007328efb8e u888888888788
Rons-Excellent-Macbook-Pro:tail rminnich$ cat /etc/hosts | ./tail -f

# serial over lan to rome
10.0.1.128 sol

192.168.42.1 milkv mv

192.168.86.219 relays
192.168.86.250 psz pistarzero

192.168.0.240 i11 u0007328efb8e u888888888788
```

same output but, again, what are you running on?

I tried this:
```
Rons-Excellent-Macbook-Pro:tail rminnich$ tail -f ../../../README.md
Improving existing commands (e.g., additional currently unsupported flags) is
very welcome. In this case it is not even required to build an initramfs, just
enter the `cmds/` directory and start coding. A list of commands that are on the
roadmap can be found [here](roadmap.md).

## Website

The sources of [u-root.org](https://u-root.org) are inside the `docs/` directory and
are deployed to the gh-pages branch. The CNAME file is currently not part of the CI
which deploys to the branch which shall be evaluated if this makes futures deployments easier.

^C
Rons-Excellent-Macbook-Pro:tail rminnich$ ./tail  -f ../../../README.md
Improving existing commands (e.g., additional currently unsupported flags) is
very welcome. In this case it is not even required to build an initramfs, just
enter the `cmds/` directory and start coding. A list of commands that are on the
roadmap can be found [here](roadmap.md).

## Website

The sources of [u-root.org](https://u-root.org) are inside the `docs/` directory and
are deployed to the gh-pages branch. The CNAME file is currently not part of the CI
which deploys to the branch which shall be evaluated if this makes futures deployments easier.
```

letian0805, 2024-10-31:
[testfile.txt](https://github.com/user-attachments/files/17580617/testfile.txt)
@rminnich You can try this file

binjip978, 2024-11-01:
Thx @letian0805 i see the problem now.

binjip978, 2024-11-02:
Number of lines is not related to this problem as I see it. But it's true that tail -f fails on any file with size more than 810 bytes. 810 is because of this:
```
// getBlockSize returns the number of bytes to read for each ReadAt call. This
// helps minimize the number of syscalls to get the last N lines of the file.
func getBlockSize(numLines uint) int64 {
	// This is currently computed as 81 * N, where N is the requested number of
	// lines, and 81 is a relatively generous estimation of the average line
	// length.
	return 81 * int64(numLines)
}
```
testcase:
```
➜  tail git:(main) ls -l f2
-rw-r--r--@ 1 binjip978  staff  810 Nov  2 08:29 f2
➜  tail git:(main) ./tail -f f2
 1 xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
 2 xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
 3 xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
 4 xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
 5 xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
 6 xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx

append one character
➜  tail git:(main) ls -l f2
-rw-r--r--@ 1 binjip978  staff  811 Nov  2 08:31 f2
➜  tail git:(main) ./tail -f f2
 2 xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
 3 xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
 4 xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
 5 xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
 6 xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxA 1 xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
 2 xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
 3 xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
 4 xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
 5 xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
 6 xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxA
 
system tail:
➜  tail git:(main) tail -f f2  
 1 xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
 2 xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
 3 xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
 4 xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
 5 xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
 6 xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxA
 ```

letian0805, 2024-11-04:
I will sort out a few cases:
1. When the text content exceeds `blkSize`, the number of lines of a `blkSize` exceeds `numLines`, which means that at most `blkSize` content will be read.
2. When the text content exceeds `blkSize`, the number of lines of a `blkSize` is less than `numLines`, but the number of text lines exceeds `numLines`, which means that more than one `blkSize` content will be read.
3. When the text content is less than or equal to `blkSize`, the number of lines exceeds `numLines`.
4. When the text content is less than or equal to `blkSize`, the number of lines is less than `numLines`.

It is obvious that the current problem is case 2. If `buf` is reused, `readData` is empty when it is read for the first time. After `readData = append(buf[:n], readData...)`, `readData` is actually `buf[:n]`. When reading again, the content of `readData` will be overwritten, resulting in data loss. This is the first problem. The second problem is that during the last read, `thisChunkSize` may be smaller than `blkSize`, and `n` returned by `ReadAt` may be between `thisChunkSize` and `blkSize`. If `buf[:n]` is used to get data, some duplicate data will be obtained.

letian0805, 2024-11-04:
To sum up, there are two problems:
1. The content read for the first time is overwritten by the content read for the second time
2. The content read for the last time may get the content read for the second to last time.

letian0805, 2024-11-04:
Let me give you an example where two problems occur at the same time: the text content is "123456789", `blkSize` is 6
First read: `readData` is empty, `buf` is "456789"
Second read: `readData` is "456789", because blkSize is 6, so it will not only read "123", `buf` is "123456".
Result: After the second read, `readData` is overwritten by "12345", and after `readData` is appended to `buf`, the final `readData` is "123456123456".

The cause of this problem, I guess, is that the code of `readLastLinesFromBeginning` was copied and simply modified.

closed 2024-12-03 by binjip978