u-root/u-root #3185: parse error for `sudo ./tcpdump -i eno1 ether broadcast`

10000TB, 2024-11-18:
**Describe the bug**
A clear and concise description of what the bug is.

`./tcpdump -i eno1 ether broadcast` does not work, complaining parsing error.

**To Reproduce**
Steps to reproduce the behavior:

```
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root/cmds/exp/tcpdump$ sudo ./tcpdump -i eno1 ether broadcast
2024/11/17 22:55:10 tcpdump: failed to compile filter into instructions: parse error
```

**Expected behavior**
A clear and concise description of what you expected to happen.

Here is what expected output look like from C tcpdump

```
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root/cmds/exp/tcpdump$ sudo tcpdump -i eno1 ether broadcast
tcpdump: verbose output suppressed, use -v[v]... for full protocol decode
listening on eno1, link-type EN10MB (Ethernet), snapshot length 262144 bytes
22:55:25.109820 ARP, Request who-has mat3en-41.svl.corp.google.com tell us-svl-mat3-core1-irb-274.n.corp.google.com, length 46
22:55:25.657255 ARP, Request who-has kumarann.svl.corp.google.com tell us-svl-mat1-core1-irb-274.n.corp.google.com, length 46
22:55:25.658663 ARP, Request who-has 100.123.244.59 tell us-svl-mat1-core1-irb-274.n.corp.google.com, length 46
22:55:25.659073 ARP, Request who-has surajan0.svl.corp.google.com tell us-svl-mat1-core1-irb-274.n.corp.google.com, length 46
22:55:25.660941 ARP, Request who-has 100.123.244.95 tell us-svl-mat1-core1-irb-274.n.corp.google.com, length 46
22:55:25.663698 ARP, Request who-has 100.123.244.59 tell us-svl-mat3-core1-irb-274.n.corp.google.com, length 46
22:55:25.663708 ARP, Request who-has hmurthy.svl.corp.google.com tell us-svl-mat3-core1-irb-274.n.corp.google.com, length 46
22:55:25.663710 ARP, Request who-has 100.123.245.108 tell us-svl-mat3-core1-irb-274.n.corp.google.com, length 46
22:55:25.663712 ARP, Request who-has surajan0.svl.corp.google.com tell us-svl-mat3-core1-irb-274.n.corp.google.com, length 46
22:55:25.666328 ARP, Request who-has xjwu.svl.corp.google.com tell us-svl-mat3-core1-irb-274.n.corp.google.com, length 46
22:55:25.666674 ARP, Request who-has 100.123.244.163 tell us-svl-mat1-core1-irb-274.n.corp.google.com, length 46
22:55:25.700175 ARP, Request who-has 100.123.244.143 tell us-svl-mat3-core1-irb-274.n.corp.google.com, length 46
22:55:25.700186 ARP, Request who-has 100.123.244.163 tell us-svl-mat3-core1-irb-274.n.corp.google.com, length 46
22:55:25.700188 ARP, Request who-has kumarann.svl.corp.google.com tell us-svl-mat3-core1-irb-274.n.corp.google.com, length 46
22:55:25.700220 ARP, Request who-has 100.123.244.95 tell us-svl-mat3-core1-irb-274.n.corp.google.com, length 46
22:55:26.191840 ARP, Request who-has hmurthy.svl.corp.google.com tell us-svl-mat3-core1-irb-274.n.corp.google.com, length 46
22:55:26.259808 ARP, Request who-has surajan0.svl.corp.google.com tell us-svl-mat1-core1-irb-274.n.corp.google.com, length 46
22:55:26.259819 ARP, Request who-has kumarann.svl.corp.google.com tell us-svl-mat1-core1-irb-274.n.corp.google.com, length 46
22:55:26.291040 ARP, Request who-has kumarann.svl.corp.google.com tell us-svl-mat3-core1-irb-274.n.corp.google.com, length 46
22:55:26.291085 ARP, Request who-has 100.123.244.163 tell us-svl-mat3-core1-irb-274.n.corp.google.com, length 46
22:55:26.291088 ARP, Request who-has 100.123.244.143 tell us-svl-mat3-core1-irb-274.n.corp.google.com, length 46
22:55:26.291106 ARP, Request who-has 100.123.244.59 tell us-svl-mat3-core1-irb-274.n.corp.google.com, length 46
22:55:26.360644 ARP, Request who-has 100.123.244.95 tell us-svl-mat1-core1-irb-274.n.corp.google.com, length 46
22:55:26.391535 ARP, Request who-has xjwu.svl.corp.google.com tell us-svl-mat3-core1-irb-274.n.corp.google.com, length 46
...
...
```

**Additional context**
Add any other context about the problem here.

RiSKeD, 2024-11-19:
The command uses the library https://github.com/packetcap/go-pcap for the parsing of expressions. 
Improvements to the filter need to be made there.

10000TB, 2024-11-19:
Fabian, could you help make the change __ there __ ? --- We review by functionality, not by changes only made to u-root repo.

jensdrenhaus, 2024-11-20:
The goal with PR #3059 was to implement a tracing tool for inbound and outbound packets against a specific
interface in native golang.
So we came up with a tcpdump-style command to bring package tracing functionality in. It was not the intention to implement a C equivalent. 
The implementation also has some limitations, as described in the PR:

> The underlying gopacket library uses c-binding, hence we use the go-pcap pkg as well for a pure native go implementation of libpcap (not all filters are supported...)

It was clear from the proposal in the PR that tcpdump here will not have 100% functionality. 
That is what you encounter here. So with this background I suggest not to label it as a bug, but a feature request.