u-root/u-root #3188: ./ip address replace ends up adding new address to a link instead of replacing

10000TB, 2024-11-20:
**Describe the bug**
A clear and concise description of what the bug is.

cmds/core/ip ends up adds new addresses to a link, instead of replacing current one.

**To Reproduce**
Steps to reproduce the behavior:

`cd cmds/core/ip/`, then run `go build .` to build ip command.

Then run following steps,

```
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root/cmds/core/ip$ ./ip address show dev testlink
7: testlink: <UP,BROADCAST,MULTICAST> mtu 1500 state DOWN group default
    link/ether 9e:59:89:6e:2a:6b
    inet 192.168.22.22 brd 192.168.22.255 scope global testlink
       valid_lft forever preferred_lft forever
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root/cmds/core/ip$ sudo ./ip address replace 192.168.22.0/12 dev testlink
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root/cmds/core/ip$ ./ip address show dev testlink
7: testlink: <UP,BROADCAST,MULTICAST> mtu 1500 state DOWN group default
    link/ether 9e:59:89:6e:2a:6b
    inet 192.168.22.22 brd 192.168.22.255 scope global testlink
       valid_lft forever preferred_lft forever
    inet 192.168.22.0 brd 192.175.255.255 scope global testlink
       valid_lft forever preferred_lft forever
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root/cmds/core/ip$ ./ip address show dev testlink
7: testlink: <UP,BROADCAST,MULTICAST> mtu 1500 state DOWN group default
    link/ether 9e:59:89:6e:2a:6b
    inet 192.168.22.22 brd 192.168.22.255 scope global testlink
       valid_lft forever preferred_lft forever
    inet 192.168.22.0 brd 192.175.255.255 scope global testlink
       valid_lft forever preferred_lft forever
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root/cmds/core/ip$ sudo ./ip address add 192.168.22.20/24 dev testlink
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root/cmds/core/ip$ ./ip address show dev testlink
7: testlink: <UP,BROADCAST,MULTICAST> mtu 1500 state DOWN group default
    link/ether 9e:59:89:6e:2a:6b
    inet 192.168.22.22 brd 192.168.22.255 scope global testlink
       valid_lft forever preferred_lft forever
    inet 192.168.22.0 brd 192.175.255.255 scope global testlink
       valid_lft forever preferred_lft forever
    inet 192.168.22.20 brd 192.168.22.255 scope global testlink
       valid_lft forever preferred_lft forever
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root/cmds/core/ip$
```

**Expected behavior**
A clear and concise description of what you expected to happen.

existing ip address should be repalced with the new address

**Additional context**
Add any other context about the problem here.

jensdrenhaus, 2024-11-20:
Taking a look on it

RiSKeD, 2024-11-20:
To my knowledge this is similar to the original IP command: 
```
ip addr show dev Test
144: Test: <BROADCAST,NOARP> mtu 4000 qdisc noop state DOWN group default 
    link/ether 56:ef:8e:34:50:04 brd ff:ff:ff:ff:ff:ff

sudo ip addr add 192.0.0.2/24 dev Test

sudo ip addr replace 192.0.0.3/24 dev Test

ip addr show dev Test
144: Test: <BROADCAST,NOARP> mtu 4000 qdisc noop state DOWN group default 
    link/ether 56:ef:8e:34:50:04 brd ff:ff:ff:ff:ff:ff
    inet 192.0.0.2/24 scope global Test
       valid_lft forever preferred_lft forever
    inet 192.0.0.3/24 scope global secondary Test
       valid_lft forever preferred_lft forever
```

jensdrenhaus, 2024-12-06:
In this case, I would not consider this a bug. 
Reopen, if there is the need for doing it differently compared to the original implementation.

closed 2024-12-06 by jensdrenhaus