u-root/u-root #3189: `ip address flush` does clear addresses of an interface

10000TB, 2024-11-20:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:

cd into `cmds/core/ip`, build the command, `go build .`, then

```
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root/cmds/core/ip$ ./ip address show dev testlink
7: testlink: <UP,BROADCAST,MULTICAST> mtu 1500 state DOWN group default
    link/ether 9e:59:89:6e:2a:6b
    inet 192.168.22.22 brd 192.168.22.255 scope global testlink
       valid_lft forever preferred_lft forever
    inet 192.168.22.0 brd 192.175.255.255 scope global testlink
       valid_lft forever preferred_lft forever
    inet 192.168.22.20 brd 192.168.22.255 scope global testlink
       valid_lft forever preferred_lft forever
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root/cmds/core/ip$ sudo ./ip link set testlink down
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root/cmds/core/ip$ ./ip address show dev testlink
7: testlink: <BROADCAST,MULTICAST> mtu 1500 state DOWN group default
    link/ether 9e:59:89:6e:2a:6b
    inet 192.168.22.22 brd 192.168.22.255 scope global testlink
       valid_lft forever preferred_lft forever
    inet 192.168.22.0 brd 192.175.255.255 scope global testlink
       valid_lft forever preferred_lft forever
    inet 192.168.22.20 brd 192.168.22.255 scope global testlink
       valid_lft forever preferred_lft forever
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root/cmds/core/ip$ ./ip address flush dev testlink
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root/cmds/core/ip$ ./ip address show dev testlink
7: testlink: <BROADCAST,MULTICAST> mtu 1500 state DOWN group default
    link/ether 9e:59:89:6e:2a:6b
    inet 192.168.22.22 brd 192.168.22.255 scope global testlink
       valid_lft forever preferred_lft forever
    inet 192.168.22.0 brd 192.175.255.255 scope global testlink
       valid_lft forever preferred_lft forever
    inet 192.168.22.20 brd 192.168.22.255 scope global testlink
       valid_lft forever preferred_lft forever
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root/cmds/core/ip$ sudo ./ip address flush dev testlink
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root/cmds/core/ip$ ./ip address show dev testlink
7: testlink: <BROADCAST,MULTICAST> mtu 1500 state DOWN group default
    link/ether 9e:59:89:6e:2a:6b
    inet 192.168.22.22 brd 192.168.22.255 scope global testlink
       valid_lft forever preferred_lft forever
    inet 192.168.22.0 brd 192.175.255.255 scope global testlink
       valid_lft forever preferred_lft forever
    inet 192.168.22.20 brd 192.168.22.255 scope global testlink
       valid_lft forever preferred_lft forever
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root/cmds/core/ip$ sudo ./ip link set dev testlink DOWN
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root/cmds/core/ip$ ./ip address show dev testlink
7: testlink: <BROADCAST,MULTICAST> mtu 1500 state DOWN group default
    link/ether 9e:59:89:6e:2a:6b
    inet 192.168.22.22 brd 192.168.22.255 scope global testlink
       valid_lft forever preferred_lft forever
    inet 192.168.22.0 brd 192.175.255.255 scope global testlink
       valid_lft forever preferred_lft forever
    inet 192.168.22.20 brd 192.168.22.255 scope global testlink
       valid_lft forever preferred_lft forever
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root/cmds/core/ip$ sudo ./ip address flush dev testlink
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root/cmds/core/ip$ ./ip address show dev testlink
7: testlink: <BROADCAST,MULTICAST> mtu 1500 state DOWN group default
    link/ether 9e:59:89:6e:2a:6b
    inet 192.168.22.22 brd 192.168.22.255 scope global testlink
       valid_lft forever preferred_lft forever
    inet 192.168.22.0 brd 192.175.255.255 scope global testlink
       valid_lft forever preferred_lft forever
    inet 192.168.22.20 brd 192.168.22.255 scope global testlink
       valid_lft forever preferred_lft forever
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root/cmds/core/ip$
```

**Expected behavior**
A clear and concise description of what you expected to happen.

Expect ip addresses flushed from the device.

**Additional context**
Add any other context about the problem here.

10000TB, 2024-11-20:
also tried `addr`, and with loops flags, no dice :( 

```
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root/cmds/core/ip$ sudo ./ip -l 4 address flush dev testlink
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root/cmds/core/ip$ ./ip address show dev testlink
7: testlink: <BROADCAST,MULTICAST> mtu 1500 state DOWN group default
    link/ether 9e:59:89:6e:2a:6b
    inet 192.168.22.22 brd 192.168.22.255 scope global testlink
       valid_lft forever preferred_lft forever
    inet 192.168.22.0 brd 192.175.255.255 scope global testlink
       valid_lft forever preferred_lft forever
    inet 192.168.22.20 brd 192.168.22.255 scope global testlink
       valid_lft forever preferred_lft forever
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root/cmds/core/ip$ sudo ./ip -l 4 addr flush dev testlink
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root/cmds/core/ip$ ./ip address show dev testlink
7: testlink: <BROADCAST,MULTICAST> mtu 1500 state DOWN group default
    link/ether 9e:59:89:6e:2a:6b
    inet 192.168.22.22 brd 192.168.22.255 scope global testlink
       valid_lft forever preferred_lft forever
    inet 192.168.22.0 brd 192.175.255.255 scope global testlink
       valid_lft forever preferred_lft forever
    inet 192.168.22.20 brd 192.168.22.255 scope global testlink
       valid_lft forever preferred_lft forever
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root/cmds/core/ip$
```

jensdrenhaus, 2024-11-20:
Taking a look

10000TB, 2024-11-20:
In comparison, **C implementation that is part of my local ubuntu works**

```
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root/cmds/core/ip$
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root/cmds/core/ip$
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root/cmds/core/ip$ ip addr show dev testlink
7: testlink: <BROADCAST,MULTICAST,MASTER> mtu 1500 qdisc noqueue state DOWN group default
    link/ether 9e:59:89:6e:2a:6b brd ff:ff:ff:ff:ff:ff
    inet 192.168.22.22/24 brd 192.168.22.255 scope global testlink
       valid_lft forever preferred_lft forever
    inet 192.168.22.0/12 brd 192.175.255.255 scope global testlink
       valid_lft forever preferred_lft forever
    inet 192.168.22.20/24 brd 192.168.22.255 scope global secondary testlink
       valid_lft forever preferred_lft forever
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root/cmds/core/ip$ ip addr flush dev testlink
Failed to send flush request: Operation not permitted
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root/cmds/core/ip$ sudo ip addr flush dev testlink
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root/cmds/core/ip$ ip addr show dev testlink
7: testlink: <BROADCAST,MULTICAST,MASTER> mtu 1500 qdisc noqueue state DOWN group default
    link/ether 9e:59:89:6e:2a:6b brd ff:ff:ff:ff:ff:ff
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root/cmds/core/ip$
```

PR #3197 (2024-11-21): cmds/core/ip: fix parsing of ip addr flush
Under certain conditions, the addresses of a device to be filtered were skipped because the function that checks the filter of the command had erroneous checks. 
Working example with the fix:  
```
❯ ./ip addr show dev Test
144: Test: <BROADCAST> mtu 4000 state DOWN group default
    link/ether 56:ef:8e:34:50:04

cmds/core/ip [ main][$!][🐹 v1.23.0][☁️  (eu-central-1)][☁️  fabian.wienand@9elements.com]
➜ sudo ./ip addr add 192.0.0.2/24 dev Test

cmds/core/ip [ main][$!][🐹 v1.23.0][☁️  (eu-central-1)][☁️  fabian.wienand@9elements.com]
➜ ./ip addr show [... cut]
closed 2024-11-21 by RiSKeD