u-root/u-root #3191: Various attributes are missing in comparision to the ip command from core utils

10000TB, 2024-11-20:
(1) It looks like a couple of attributes are missing in comparision to the ip command from core utils __ from my end __ -- is your test log from the golang command or c implementatio on your linux ?

(a) qdisc noqueue -- is missing
(b) mode DEFAULT -- is missing
(c) qlen 1000 -- is missing
(d) MAC for link / loopback is missing.
(e) brd  -- info is missing, e.g. 

link/ether 1c:a0:b8:77:f4:22

while we should have 

link/ether 1c:a0:b8:77:f4:22 brd ff:ff:ff:ff:ff:ff

```
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root/cmds/core/ip$ ip link
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN mode DEFAULT group default qlen 1000
link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
2: eno1: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP mode DEFAULT group default qlen 1000
link/ether 1c:a0:b8:77:f4:22 brd ff:ff:ff:ff:ff:ff
altname enp0s31f6
4: docker0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue state DOWN mode DEFAULT group default
link/ether 02:42:e4:13:32:83 brd ff:ff:ff:ff:ff:ff
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root/cmds/core/ip$ ./ip link
1: lo: <UP,LOOPBACK> mtu 65536 state UNKNOWN group default
link/loopback
2: eno1: <UP,BROADCAST,MULTICAST> mtu 1500 state UP group default
link/ether 1c:a0:b8:77:f4:22
4: docker0: <UP,BROADCAST,MULTICAST> mtu 1500 state DOWN group default
link/ether 02:42:e4:13:32:83
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root/cmds/core/ip$
```

10000TB, 2024-11-20:
@RiSKeD : "Yes that is correct. Certain information is not accessible via our external library depency and cannot be displayed because of this."

10000TB, 2024-11-20:
I am ok with with the explanation for this MVP implementation.

Thank you Fabian

10000TB, 2024-11-20:
on a second thought, I'd like to keep this open to track the improvement.

jensdrenhaus, 2024-12-06:
> on a second thought, I'd like to keep this open to track the improvement.

Maybe this could be relabeled to "Feature Request" in this case.