u-root/u-root #3194: `--color / -c` is part of `ip` cmd help msg, but not supported

10000TB, 2024-11-20:
```

IP golang/tinygo (works the same) 

Commit-Hash: 53a75cace928ab03f20a34b7445ab9f881770854 

OS: Ubuntu-22.04 

 
Currently unsupported a: -r, -o, Comment start  -c Comment end  

If a user still inputs the flags, an error message is displayed. 

```

10000TB, 2024-11-20:
@RiSKeD @jensdrenhaus shared with me offline that there was good reason / considerations that led us not to implement it.

10000TB, 2024-11-20:
I shared that we should strive for a good working MVP implementation, so that explanation is aligned. I am ok with it.

10000TB, 2024-11-20:
@RiSKeD could you elaborate the trade-offs / challenges encountered ?

RiSKeD, 2024-11-20:
Certainly, the flag was dropped as it would require at least an external library or our own implementation to display coloured text. 
As we want to stay small binary-wise, I opted to drop the flag instead of introducing additional binary size to the command.

10000TB, 2024-12-19:
SG, I agree coloring can be waived from MVP requirements.

closed 2024-12-19 by 10000TB