u-root/u-root #3195: `-o` option not supported in `cmds/core/ip` cmd.

10000TB, 2024-11-20:
```
➜ ./ip Comment start  -ao Comment end  addr show 

2024/11/14 12:24:51 ip: outputting each record on a single line is unsupported 

```

I was wondering IF this option is not supported, any reason why we still keep it as part of help message ?

@RiSKeD / @jensdrenhaus shared there was challenges in implementing in the MVP phase.

10000TB, 2024-11-20:
@RiSKeD could you elaborate the challenges / considerations that went in the decision on not implementing  this as part of MVP delivery ?

RiSKeD, 2024-11-20:
Of course, this option was dropped as priority was shifted towards more crucial flags inside the IP command. There are no challenges per se implementation-wise, just more of a prioritization for the MVP delivery.

jensdrenhaus, 2024-12-06:
This is a low-hanging fruit to add on the way forward.

closed 2025-02-11 by jensdrenhaus