u-root/u-root #3201: `cmds/core/ip route` does not show routes, but saying invalid argument

10000TB, 2024-11-20:
**Describe the bug**
A clear and concise description of what the bug is.

`cmds/core/ip route` does not show routes, but saying invalid argument

**To Reproduce**
Steps to reproduce the behavior:

```
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root/cmds/core/ip$ ./ip route
2024/11/20 04:30:16 ip: invalid argument
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root/cmds/core/ip$
```

**Expected behavior**
A clear and concise description of what you expected to happen.

Expect it to show / list routes, like C lang ip command, 

```
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root/cmds/core/ip$ ip route
default via 100.123.245.254 dev eno1 proto dhcp src 100.123.244.123 metric 100
100.123.244.0/23 dev eno1 proto kernel scope link src 100.123.244.123 metric 100
unreachable 169.254.169.254
192.168.9.0/24 dev docker0 proto kernel scope link src 192.168.9.1 linkdown
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root/cmds/core/ip$
```

**Additional context**
Add any other context about the problem here.

10000TB, 2024-11-20:
`./ip route list` also broke

```
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root/cmds/core/ip$ ./ip route list
2024/11/20 04:35:40 ip: invalid argument
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root/cmds/core/ip$
```

10000TB, 2024-11-20:
Can we also add a test case to safeguard against regression for future ?

jensdrenhaus, 2024-11-20:
this works for me, too 
```
jens@X1:~/projects/u-root/u-root/cmds/core/ip(main)$ git checkout 81ba4a72f49f4d71db22a1f9e43d2522cb097892
Note: switching to '81ba4a72f49f4d71db22a1f9e43d2522cb097892'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

HEAD is now at 81ba4a72 Fix brctl show functionality
jens@X1:~/projects/u-root/u-root/cmds/core/ip((HEAD detached at 81ba4a72))$ pwd
/home/jens/projects/u-root/u-root/cmds/core/ip
jens@X1:~/projects/u-root/u-root/cmds/core/ip((HEAD detached at 81ba4a72))$ go build
jens@X1:~/projects/u-root/u-root/cmds/core/ip((HEAD detached at 81ba4a72))$ ./ip route
default via 10.93.128.1 dev enxa0cec8be1bd6 proto dhcp metric 100
default via 10.93.12.1 dev wlp0s20f3 proto dhcp metric 600
10.93.12.0/22 dev wlp0s20f3 proto kernel scope link src 10.93.13.88 metric 600
10.93.128.0/23 dev enxa0cec8be1bd6 proto kernel scope link src 10.93.128.142 metric 100
jens@X1:~/projects/u-root/u-root/cmds/core/ip((HEAD detached at 81ba4a72))$ ./ip route list
default via 10.93.128.1 dev enxa0cec8be1bd6 proto dhcp metric 100
default via 10.93.12.1 dev wlp0s20f3 proto dhcp metric 600
10.93.12.0/22 dev wlp0s20f3 proto kernel scope link src 10.93.13.88 metric 600
10.93.128.0/23 dev enxa0cec8be1bd6 proto kernel scope link src 10.93.128.142 metric 100
jens@X1:~/projects/u-root/u-root/cmds/core/ip((HEAD detached at 81ba4a72))$ 
```

10000TB, 2024-11-20:
trying again

10000TB, 2024-11-20:
> @10000TB Interesting, this is not reproducible for me. Can you provide more system details? I am running the command with Ubuntu 22.04 on x86 with commit [9939f25](https://github.com/u-root/u-root/commit/9939f258f11a39fbd3d9c3eb7f4d3fda080fd7dd).
> 
> ```
> Commit: 9939f258 (HEAD -> main, origin/main, origin/HEAD) cmds/core/ip: tcpmetrics, show all ip family metrics if none specific selected
> 
> ➜ ./ip route
> default via 10.93.128.1 dev enxa0cec8be26d0 proto dhcp metric 100
> default via 10.93.128.1 dev wlp3s0 proto dhcp metric 600
> 10.93.128.0/23 dev enxa0cec8be26d0 proto kernel scope link src 10.93.128.171 metric 100
> 10.93.128.0/23 dev wlp3s0 proto kernel scope link src 10.93.128.215 metric 600
> 169.254.0.0/16 dev wlp3s0 proto boot scope link src <nil> metric 1000
> 172.17.0.0/16 dev docker0 proto kernel scope link src 172.17.0.1 metric 0
> 172.18.0.0/16 dev br-4280b4c9b93d proto kernel scope link src 172.18.0.1 metric 0
> 172.24.0.0/16 dev br-53e683fe9d03 proto kernel scope link src 172.24.0.1 metric 0
> 172.29.0.0/16 dev br-f907b14dae5f proto kernel scope link src 172.29.0.1 metric 0
> 192.168.144.0/20 dev br-e7c1d281f747 proto kernel scope link src 192.168.144.1 metric 0
> 
> ➜ ./ip route list
> default via 10.93.128.1 dev enxa0cec8be26d0 proto dhcp metric 100
> default via 10.93.128.1 dev wlp3s0 proto dhcp metric 600
> 10.93.128.0/23 dev enxa0cec8be26d0 proto kernel scope link src 10.93.128.171 metric 100
> 10.93.128.0/23 dev wlp3s0 proto kernel scope link src 10.93.128.215 metric 600
> 169.254.0.0/16 dev wlp3s0 proto boot scope link src <nil> metric 1000
> 172.17.0.0/16 dev docker0 proto kernel scope link src 172.17.0.1 metric 0
> 172.18.0.0/16 dev br-4280b4c9b93d proto kernel scope link src 172.18.0.1 metric 0
> 172.24.0.0/16 dev br-53e683fe9d03 proto kernel scope link src 172.24.0.1 metric 0
> 172.29.0.0/16 dev br-f907b14dae5f proto kernel scope link src 172.29.0.1 metric 0
> 192.168.144.0/20 dev br-e7c1d281f747 proto kernel scope link src 192.168.144.1 metric 0
> ```

It looks like commit you are on are from last week.

10000TB, 2024-11-20:
Could you try rebase onto HEAD / main branch, and re-try ? -- This looks like a regression.

jensdrenhaus, 2024-11-20:
Could you also print the output of `go env` pls.

10000TB, 2024-11-24:
```
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root/cmds/core/ip$ cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux rodete"
NAME="Debian GNU/Linux rodete"
VERSION_CODENAME=rodete
ID=debian
HOME_URL=<redacted>
SUPPORT_URL=<redacted>
BUG_REPORT_URL=<redacted>
xuehaohu@worldpeace:
```

10000TB, 2024-11-24:
```
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root/cmds/core/ip$ go env
GO111MODULE=''
GOARCH='amd64'
GOBIN=''
GOCACHE='/usr/local/google/home/xuehaohu/.cache/go-build'
GOENV='/usr/local/google/home/xuehaohu/.config/go/env'
GOEXE=''
GOEXPERIMENT='fieldtrack,boringcrypto'
GOFLAGS=''
GOHOSTARCH='amd64'
GOHOSTOS='linux'
GOINSECURE=''
GOMODCACHE='/usr/local/google/home/xuehaohu/go/pkg/mod'
GONOPROXY=''
GONOSUMDB=''
GOOS='linux'
GOPATH='/usr/local/google/home/xuehaohu/go'
GOPRIVATE=''
GOPROXY='https://proxy.golang.org,direct'
GOROOT='/usr/lib/google-golang'
GOSUMDB='sum.golang.org'
GOTMPDIR=''
GOTOOLCHAIN='auto'
GOTOOLDIR='/usr/lib/google-golang/pkg/tool/linux_amd64'
GOVCS=''
GOVERSION='go1.23-20240626-RC01 cl/646990413 +5a18e79687 X:fieldtrack,boringcrypto'
GODEBUG=''
GOTELEMETRY='local'
GOTELEMETRYDIR='/usr/local/google/home/xuehaohu/.config/go/telemetry'
GCCGO='gccgo'
GOAMD64='v1'
AR='ar'
CC='clang'
CXX='clang++'
CGO_ENABLED='1'
GOMOD='/usr/local/google/home/xuehaohu/go/src/github.com/u-root/u-root/go.mod'
GOWORK=''
CGO_CFLAGS='-O2 -g'
CGO_CPPFLAGS=''
CGO_CXXFLAGS='-O2 -g'
CGO_FFLAGS='-O2 -g'
CGO_LDFLAGS='-O2 -g'
PKG_CONFIG='pkg-config'
GOGCCFLAGS='-fPIC -m64 -pthread -fno-caret-diagnostics -Qunused-arguments -Wl,--no-gc-sections -fmessage-length=0 -ffile-prefix-map=/tmp/go-build2628181808=/tmp/go-build -gno-record-gcc-switches'
```

leongross, 2024-12-02:
@10000TB I launched a docker container based on `debian:testing`, installed `go1.23`, built the `ip` command, and ran it, but I could not reproduce the error. What kernel version does your system run? What is the output of `uname -a`?

PR #3301 (2025-02-13): Fix ip route
This is a fix for calling 'ip route' without further arguments. It now results in 'ip route show' as expected.
Fixes #3201