u-root/u-root #3205: `./brctl showmacs <bdg>`: `open /sys/class/net/testbridge/brforward: no such file or directory`

10000TB, 2024-11-24:
**Describe the bug**

`showmacs` does not work. **Does it need any kernel config be enabled ?**

**To Reproduce**
Steps to reproduce the behavior:

```
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root/cmds/core/brctl$
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root/cmds/core/brctl$ sudo ./brctl showmacs testbridge
2024/11/24 07:20:00 brctl: Readfile("/sys/class/net/testbridge/brforward"): open /sys/class/net/testbridge/brforward: no such file or directory
```

**Expected behavior**
A clear and concise description of what you expected to happen.

```
sudo ./brctl showmacs dummyBridge
port no	mac addr    	is_local?
257	6e:66:91:cc:6f:4f	false
257	6e:66:91:cc:6f:4f	false
```

**Additional context**
Add any other context about the problem here.

ChriMarMe, 2024-12-02:
Error is not reproducible locally. Not enough information to recreate. 

Probably `testbridge` was not added as bridge with `sudo ./brctl addbr testbridge`

jensdrenhaus, 2024-12-06:
Works for me when set up the bridg in advence like so: 
```
jens@X1:~/projects/u-root/u-root/cmds/core/brctl(main)$ git log --oneline -1
439a6abc (HEAD -> main, origin/main, origin/HEAD) remove tinygo exclusion build tag from kmoudle compression handler
jens@X1:~/projects/u-root/u-root/cmds/core/brctl(main)$ sudo ./brctl addbr testbridge
2024/12/06 01:25:21 brctl: executeIoctlStr: syscall.Syscall: file exists
jens@X1:~/projects/u-root/u-root/cmds/core/brctl(main)$ ./brctl showmacs testbridge
port no	mac addr		is_local?
jens@X1:~/projects/u-root/u-root/cmds/core/brctl(main)$ sudo ip link add dev testdev up type dummy
jens@X1:~/projects/u-root/u-root/cmds/core/brctl(main)$ sudo ./brctl addif testbridge testdev
jens@X1:~/projects/u-root/u-root/cmds/core/brctl(main)$ sudo ./brctl show testbridge
bridge name                   bridge id     STP enabled           interfaces
testbridge            8000.8e8592fd2801           false             testdev 
jens@X1:~/projects/u-root/u-root/cmds/core/brctl(main)$ ./brctl showmacs testbridge
port no	mac addr		is_local?
257	de:c1:7d:ca:f6:26	false
257	de:c1:7d:ca:f6:26	false
jens@X1:~/projects/u-root/u-root/cmds/core/brctl(main)$
```

Can you confirm, that you did the same before calling `./brctl showmacs` ? 
Further, in your failing example you have "testbridge" whereas in your expected text you wrote "dummyBridge" - did you pass the right one to `./brctl showmacs` ?

jensdrenhaus, 2024-12-18:
@10000TB can you confirm?

closed 2025-04-05 by jensdrenhaus