u-root/u-root #3206: `./netcat -l -p <port>` does not take port number passed in

10000TB, 2024-11-24:
**Describe the bug**
A clear and concise description of what the bug is.

port number is not taken in.

`./netcat` built from this commit

```
commit b2007a3a212cce4e6b2461bdf58e346e87aeefb2 (HEAD -> main, origin/main, origin/HEAD)
Author: Ronald G Minnich <rminnich@gmail.com>
Date:   Fri Nov 22 14:41:15 2024 -0800

    cmds/dd: remove unused flags from dd func

    Signed-off-by: Ronald G Minnich <rminnich@gmail.com>
```

**To Reproduce**
Steps to reproduce the behavior:

```
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root/cmds/core/netcat$ ./netcat -l -p "31336"
2024/11/24 07:45:11 error: failed to setup listener: listen tcp 0.0.0.0:31337: bind: address already in use
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root/cmds/core/netcat$ ./netcat -p 31337 -l
2024/11/24 07:45:26 error: failed to setup listener: listen tcp 0.0.0.0:31337: bind: address already in use
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root/cmds/core/netcat$ ./netcat -p 31117 -l
2024/11/24 07:45:33 error: failed to setup listener: listen tcp 0.0.0.0:31337: bind: address already in use
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root/cmds/core/netcat$ ./netcat -p 111 -l
2024/11/24 07:45:39 error: failed to setup listener: listen tcp 0.0.0.0:31337: bind: address already in use
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root/cmds/core/netcat$ sudo ./netcat -p 111 -l
2024/11/24 07:45:48 error: failed to setup listener: listen tcp 0.0.0.0:31337: bind: address already in use
```

**Expected behavior**
A clear and concise description of what you expected to happen.

port number passed in is taken, and listening on the specified port.

**Additional context**
Add any other context about the problem here.

10000TB, 2024-11-25:
`./netcat -l localhost 8080` did seem to work,

RiSKeD, 2024-11-25:
As --source is not supported, I thought the same would apply to source-port, but this is not true In the original command. Thanks for pointing it out, working on a fix.

PR #3208 (2024-12-13): cmds/core/netcat: fix parsing of source-port flag
resolves #3206 
The source port flag was not correctly parsed in listen Mode.

Working examples: 
```
➜ ./netcat -l -p 3030 localhost
test
Second shell:
➜ ./netcat -C localhost 3030
2024/11/25 08:54:56 Connection to localhost:3030 [tcp] succeeded
test
```

```
❯ ./netcat -l localhost 7654
test
Second shell:
❯ ./netcat -C localhost 7654
2024/11/25 08:57:05 Connection to localhost:7654 [tcp] succeeded
test
```
Fail with 2 specified ports:
```
❯ ./netcat -l -p 3222 localhost 7654
2024/11/25 08:57:57 error: got more than one port specification
```
 [... cut]
closed 2024-12-13 by RiSKeD