u-root/u-root #3213: `./netcat -l --udp localhost 8080` fail to write connection - destination address required

10000TB, 2024-11-30:
**Describe the bug**
A clear and concise description of what the bug is.

Following error were observed from running netcat command to start listening on 8080, and then there is an error complaining "failed to write to connection: ..."

```
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root/cmds/core/netcat$ ./netcat -l --udp localhost 8080

2024/11/30 08:53:41 failed to write to connection: write udp 127.0.0.1:8080: write: destination address required

```

**To Reproduce**
Steps to reproduce the behavior:

` ./netcat -l --udp localhost 8080`

**Expected behavior**
A clear and concise description of what you expected to happen.

no confusing error. 

**Additional context**
Add any other context about the problem here.

RiSKeD, 2024-12-02:
The command works for me on Ubuntu 22.04: 
```
❯ ./netcat -l --udp localhost 8082
abc
TestData
^C
Second shell:
❯ ./netcat -C --udp localhost 8082
2024/12/02 10:31:11 Connection to localhost:8082 [udp] succeeded
abc
TestData
^C
```
Maybe this is something configuration-specific?

RiSKeD, 2024-12-02:
I've confirmed, this does not work with an debian:testing docker image.

RiSKeD, 2024-12-09:
So i've narrowed the issue down on debian-testing: 
```
root@03f152d19306:/tmp/u-root/cmds/core/netcat# ./netcat --udp -l localhost 8080
``` 
will open up a listener on 127.0.0.1:8080. 
```
root@03f152d19306:/tmp/u-root/cmds/core/netcat# ss -tuln
Netid               State                Recv-Q               Send-Q                             Local Address:Port                             Peer Address:Port               
udp                 UNCONN               0                    0                                      127.0.0.1:8080                                  0.0.0.0:*           
```
With the original command, this differs in a listener on [::1]:8080.
The original command and u-root command will try to connect to [::1]:8080 if the localhost is passed in connection mode.
If we connect via 127.0.0.1 everything works well:
```
root@03f152d19306:/tmp/u-root/cmds/core/netcat# ./netcat --udp -l localhost 8080
Test1
Test 3
Test data!
Second shell:
root@03f152d19306:/tmp/u-root/cmds/core/netcat# ./netcat --udp -C 127.0.0.1 8080
2024/12/09 12:31:16 Connection to 127.0.0.1:8080 [udp] succeeded
Test1
Test 3
Test data!
```
So the issue lies in the resolving of localhost, which somehow maps to 127.0.0.1 in listen mode and  [::1] in connect mode.

RiSKeD, 2024-12-09:
So the resolving of localhost on listening side is inside the net pkg:
```
conn, err = net.ListenUDP(network, udpAddr)
```
And the resolving of localhost on the connection side also executes via the net pkg:
```
conn, err = dialer.Dial(network, address)
```
Strange this would result both in ipv4 and ipv6.

RiSKeD, 2024-12-09:
@10000TB Can you check if this works for you as well? 
```
root@03f152d19306:/tmp/u-root/cmds/core/netcat# ./netcat --udp -l localhost 8080
Test1
Test 3
Test data!
Second shell:
root@03f152d19306:/tmp/u-root/cmds/core/netcat# ./netcat --udp -C 127.0.0.1 8080
2024/12/09 12:31:16 Connection to 127.0.0.1:8080 [udp] succeeded
Test1
Test 3
Test data!
```
This seems like a problem with DNS resolution inside the net pkg rather than the tool itself.

jensdrenhaus, 2025-01-30:
So it seems that the Golang net package behaves not quite consistent if you are not specifying IPv4 or IPv6. This leads to problems especially when using UDP. See https://gist.github.com/jensdrenhaus/8ce19554fbbe57417388305e1f69bef3

Taking this, I'll adapt the netcat tool, to do specific requests in all situations.

PR #3291 (2025-03-19): netcat: explicit use of IPv4 and IPv6
Fixes #3213
closed 2025-03-19 by jensdrenhaus