u-root/u-root #3215: `./netcat -C --exec` executing the given bash cmd reports extra write failure, ` failed to write to output`

10000TB, 2024-12-01:
**Describe the bug**
A clear and concise description of what the bug is.

(shell1)

```
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root/cmds/core/netcat$
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root/cmds/core/netcat$ ./netcat -l localhost 8080

Hello.
2024/12/01 08:12:22 failed to write to output: writeto tcp 127.0.0.1:8080->127.0.0.1:44224: read tcp 127.0.0.1:8080->127.0.0.1:44224: read: connection reset by peer

```

(shell2)

```
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root/cmds/core/netcat$ ./netcat -C --exec "/bin/echo Hello." localhost 8080
2024/12/01 08:12:22 Connection to localhost:8080 [tcp] succeeded
```

**To Reproduce**
Steps to reproduce the behavior:

see commands from above.

**Expected behavior**
A clear and concise description of what you expected to happen.

```
// Run in first shell
./netcat -l localhost 8080 

// Run in second shell:
./netcat -C --exec "/bin/echo Hello." localhost 8080
2024/09/09 08:50:40 Connection to localhost:8080 [tcp] succeeded

// Output in first shell (The command is run in second shell and output is // send to first shell)
Hello.
```

**Additional context**
Add any other context about the problem here.

RiSKeD, 2024-12-02:
Maybe there is something configuration-specific here as well. With Ubuntu 22.04 it works fine:
```
❯ ./netcat -l localhost 8080
Hello.

Second shell:
➜ ncat -C --exec "/bin/echo Hello." localhost 8080
```

RiSKeD, 2024-12-02:
With Ubuntu and Commit: e3b3cae2 (origin/HEAD) it works for me fine:
```
➜ ./netcat -l localhost 4444
Hello.
➜ ./netcat -C --exec "/bin/echo Hello." localhost 4444
2024/12/02 10:52:50 Connection to localhost:4444 [tcp] succeeded
```

RiSKeD, 2024-12-02:
Also works on debian:testing:
```
root@5e8edba6f31d:~/u-root/cmds/core/netcat# cat /etc/os-release 
PRETTY_NAME="Debian GNU/Linux trixie/sid"
NAME="Debian GNU/Linux"
VERSION_CODENAME=trixie
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
root@5e8edba6f31d:~/u-root/cmds/core/netcat#  ./netcat -l localhost 4444
Hello.
Second shell:
root@5e8edba6f31d:~/u-root/cmds/core/netcat# ./netcat -C --exec "/bin/echo Hello." localhost 4444
2024/12/02 10:06:26 Connection to localhost:4444 [tcp] succeeded
```

jensdrenhaus, 2024-12-06:
Works for me, too:
first shell
```
jens@X1:~/projects/u-root/u-root/cmds/core/netcat(main)$ git log --oneline -1
439a6abc (HEAD -> main, origin/main, origin/HEAD) remove tinygo exclusion build tag from kmoudle compression handler
jens@X1:~/projects/u-root/u-root/cmds/core/netcat(main)$ ./netcat -l localhost 4444
Hello.
```

second shell
```
jens@X1:~/projects/u-root/u-root/cmds/core/netcat(main)$ git log --oneline -1
439a6abc (HEAD -> main, origin/main, origin/HEAD) remove tinygo exclusion build tag from kmoudle compression handler
jens@X1:~/projects/u-root/u-root/cmds/core/netcat(main)$ ./netcat -C --exec "/bin/echo Hello." localhost 4444
2024/12/06 01:13:29 Connection to localhost:4444 [tcp] succeeded
jens@X1:~/projects/u-root/u-root/cmds/core/netcat(main)$ 
```

Maybe check the used commit. Other ways, could you provide more information about your system, pls?

jensdrenhaus, 2024-12-18:
@10000TB can you check again, taking the above into account or provide more detail of your environment, pls.