u-root/u-root #3217: `./netstat -wide` - flag provided but not defined: -d

10000TB, 2024-12-01:
**Describe the bug**
A clear and concise description of what the bug is.

```
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root/cmds/core/netstat$ ./netstat -wide
flag provided but not defined: -d
Usage of ./netstat:
  -4	IPv4 fs. default: false
  -6	IPv6 fs. default: false
  -C
  -I string
...
```

**To Reproduce**
Steps to reproduce the behavior:

Build `netstat` from 

```
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root/cmds/core/netstat$ git log -1
commit e3b3cae21292a1c40d75d9066b1aeec446f790ea (HEAD -> main, origin/main, origin/HEAD)
Author: Ronald G Minnich <rminnich@gmail.com>
Date:   Sat Nov 30 19:55:00 2024 -0800

    Fix race in gosh_test.go

    Signed-off-by: Ronald G Minnich <rminnich@gmail.com>
```

Run `$ ./netstat -wide`

**Expected behavior**
A clear and concise description of what you expected to happen.

Expect net stat info printed out.

**Additional context**
Add any other context about the problem here.

10000TB, 2024-12-01:
Perhaps we need `--wide` (two hyphens), other than `-wide` ( one hyphen ), in the help text ?

ChriMarMe, 2024-12-02:
Yes, two dashes are required to operate the command properly for the long versions of commands.

The help text is generated by the flag package. To mimic the POSIX-like flag usage, this is the compromise we had to make. Either `--longcommand` OR `-s` (s for shortcommand here).

I will adapt the printing of help.

jensdrenhaus, 2024-12-06:
Addressed #3222

closed 2024-12-06 by jensdrenhaus