u-root/u-root #3219: `./netstat -I eno1` complains `unsigned integer overflow on token 98831250889`

10000TB, 2024-12-01:
**Describe the bug**
A clear and concise description of what the bug is.

`./netstat -I eno1` complains `unsigned integer overflow on token 98831250889`

**To Reproduce**
Steps to reproduce the behavior:

HEAD commit as follows when testing. Build and run `netstat -I <local interface>`

```
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root/cmds/core/netstat$ git log -1
commit e3b3cae21292a1c40d75d9066b1aeec446f790ea (HEAD -> main, origin/main, origin/HEAD)
Author: Ronald G Minnich <rminnich@gmail.com>
Date:   Sat Nov 30 19:55:00 2024 -0800

    Fix race in gosh_test.go

    Signed-off-by: Ronald G Minnich <rminnich@gmail.com>
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root/cmds/core/netstat$
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root/cmds/core/netstat$ ./netstat -I eno1
2024/12/01 09:23:51 unsigned integer overflow on token 98831250889
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root/cmds/core/netstat$
```

**Expected behavior**
A clear and concise description of what you expected to happen.

```
./netstat -I enp7s0f3u1u2c2
Kernel Interface table
Iface        	MTU  	Rx-OK	Rx-ERR   Rx-DRP   Rx-OVR   TX-OK	TX-ERR   TX-DRP   TX-OVR   Flg
enp7s0f3u1u2c2   1500 	4894803  0    	3    	0    	1699749  0    	0    	0    	BUMR
```

**Additional context**
Add any other context about the problem here.

ChriMarMe, 2024-12-02:
I cant reproduce the error with Fedora 40 native and Debian:testing in docker.

jensdrenhaus, 2024-12-05:
I encounter the same issue, will take a look.

closed 2025-01-22 by jensdrenhaus