u-root/u-root #3223: `cmds/core/gpgv` does not build using tinygo toolchain

10000TB, 2024-12-02:
**Describe the bug**
A clear and concise description of what the bug is.

`xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root/cmds/core/gpgv$ $BASE/tinygo-org/tinygo/build/tinygo build -tags netgo,purego,noasm,tinygo.enable .` would casue error.

```
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root/cmds/core/gpgv$ $BASE/tinygo-org/tinygo/build/tinygo build -tags netgo,purego,noasm,tinygo.enable .

# github.com/ProtonMail/go-crypto/openpgp/internal/algorithm
/usr/local/google/home/xuehaohu/go/src/github.com/tinygo-org/tinygo/src/runtime/hashmap.go:474:12: interp: unknown type kind in markExternalValue
  call void @runtime.hashmapSet(ptr %m, ptr %key, ptr %value, i32 %9, ptr undef), !dbg !23122

traceback:
/usr/local/google/home/xuehaohu/go/src/github.com/tinygo-org/tinygo/src/runtime/hashmap.go:474:12:
  call void @runtime.hashmapSet(ptr %m, ptr %key, ptr %value, i32 %9, ptr undef), !dbg !23122
/usr/local/google/home/xuehaohu/go/src/github.com/u-root/u-root/vendor/github.com/ProtonMail/go-crypto/openpgp/internal/algorithm:
  call void @runtime.hashmapBinarySet(ptr %1, ptr @"github.com/ProtonMail/go-crypto/openpgp/internal/algorithm$alloca", ptr @"github.com/ProtonMail/go-crypto/openpgp/internal/algorithm$alloca.30", ptr undef), !dbg !23109
```

**To Reproduce**
Steps to reproduce the behavior:

see above.

**Expected behavior**
A clear and concise description of what you expected to happen.

The cmdlet should build, and produce an executable.

**Additional context**
Add any other context about the problem here.

10000TB, 2024-12-02:
(1) golang toolchain can build ok, 

```
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root/cmds/core/gpgv$ go build .
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root/cmds/core/gpgv$
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root/cmds/core/gpgv$ ls
gpgv  gpgv.go  gpgv_test.go  testdata
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root/cmds/core/gpgv$
```

(2) My tinygo repo setup can allow me build most other `core` cmds.

leongross, 2024-12-04:
I could not reproduce this behavior on my development machine with the following versions:

* u-root: `71f2a92292b86b46e374d2f9711f63d89bd8324f`
* tinygo: `b5a8931fb57ca93fdfcf9c4e443833cd0def2a8d`
* golang: `go version go1.23.0 linux/amd64`
* distro: `Fedora 40 Workstation edition`
* kernel: `Linux fedora-9e 6.10.7-200.fc40.x86_64`

If you can find a version mismatch compared to your setup this might help us to figure out what the problem is.

jensdrenhaus, 2024-12-05:
Works on my system with the shown commtis:
```
jens@X1:~/projects/u-root/tinygo((HEAD detached at 6593cf22))$ cd $BASE/u-root/cmds/core/gpgv
jens@X1:~/projects/u-root/u-root/cmds/core/gpgv(main)$ git log --oneline -1
439a6abc (HEAD -> main, origin/main, origin/HEAD) remove tinygo exclusion build tag from kmoudle compression handler
jens@X1:~/projects/u-root/u-root/cmds/core/gpgv(main)$ $BASE/tinygo/build/tinygo build -tags netgo,purego,noasm,tinygo.enable .
jens@X1:~/projects/u-root/u-root/cmds/core/gpgv(main)$ ls
gpgv  gpgv.go  gpgv_test.go  testdata
jens@X1:~/projects/u-root/u-root/cmds/core/gpgv(main)$ 
```

jensdrenhaus, 2024-12-18:
@10000TB have you tested again?