u-root/u-root #3225: `cmds/core/init` does not build with `tinygo` toolchain

10000TB, 2024-12-02:
**Describe the bug**

```
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root/cmds/core/init$ $BASE/tinygo-org/tinygo/build/tinygo build -tags netgo,purego,noasm,tinygo.enable .
# github.com/u-root/u-root/pkg/kmodule
../../../pkg/kmodule/kmodule_linux.go:65:9: undefined: compressionReader
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root/cmds/core/init$
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root/cmds/core/init$
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root/cmds/core/init$ $BASE/tinygo-org/tinygo/build/tinygo build -tags netgo,purego,noasm,tinygo.enable .
# github.com/u-root/u-root/pkg/kmodule
../../../pkg/kmodule/kmodule_linux.go:65:9: undefined: compressionReader
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root/cmds/core/init$ git log -1
commit e3b3cae21292a1c40d75d9066b1aeec446f790ea (HEAD -> main, origin/main, origin/HEAD)
Author: Ronald G Minnich <rminnich@gmail.com>
Date:   Sat Nov 30 19:55:00 2024 -0800

    Fix race in gosh_test.go

    Signed-off-by: Ronald G Minnich <rminnich@gmail.com>
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root/cmds/core/init$
```

**To Reproduce**
Steps to reproduce the behavior:

see above. checkout the HEAD commit, and run tinygo build.

**Expected behavior**
A clear and concise description of what you expected to happen.

The command should build w/o error.

**Additional context**
Add any other context about the problem here.

leongross, 2024-12-05:
Pull Request #3235 should have resolved this issue. That means commit 439a6abc142872bc0fe09e6299376cacec2c292f should built just fine.

jensdrenhaus, 2024-12-05:
Yes, works with more recent commit:
```
jens@X1:~/projects/u-root/tinygo((HEAD detached at 6593cf22))$ cd $BASE/tinygo/
jens@X1:~/projects/u-root/tinygo((HEAD detached at 6593cf22))$ cd $BASE/u-root/cmds/core/init/
jens@X1:~/projects/u-root/u-root/cmds/core/init(main)$ git log --oneline -1
439a6abc (HEAD -> main, origin/main, origin/HEAD) remove tinygo exclusion build tag from kmoudle compression handler
jens@X1:~/projects/u-root/u-root/cmds/core/init(main)$ $BASE/tinygo/build/tinygo build -tags netgo,purego,noasm,tinygo.enable .
jens@X1:~/projects/u-root/u-root/cmds/core/init(main)$ ls
fns_other.go  fns_unix.go  init  init_freebsd.go  init.go  init_linux.go  init_plan9.go
jens@X1:~/projects/u-root/u-root/cmds/core/init(main)$ 
```

closed 2024-12-05 by jensdrenhaus