u-root/u-root #3227: `cmds/core/kexec` does not build with `tinygo` toochain, complaining `ld.lld: error: unknown argument '-pthread'`

10000TB, 2024-12-02:
**Describe the bug**
A clear and concise description of what the bug is.

```
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root/cmds/core/kexec$
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root/cmds/core/kexec$ ls
kexec_linux.go  kexec_test.go
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root/cmds/core/kexec$ $BASE/tinygo-org/tinygo/build/tinygo build -tags netgo,purego,noasm,tinygo.enable .

ld.lld: error: unknown argument '-pthread'
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root/cmds/core/kexec$
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root/cmds/core/kexec$ ls
kexec_linux.go  kexec_test.go
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root/cmds/core/kexec$ git log -1
commit e3b3cae21292a1c40d75d9066b1aeec446f790ea (HEAD -> main, origin/main, origin/HEAD)
Author: Ronald G Minnich <rminnich@gmail.com>
Date:   Sat Nov 30 19:55:00 2024 -0800

    Fix race in gosh_test.go

    Signed-off-by: Ronald G Minnich <rminnich@gmail.com>
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root/cmds/core/kexec$
```

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
A clear and concise description of what you expected to happen.

**Additional context**
Add any other context about the problem here.

leongross, 2024-12-05:
The command builds on my development system with
* u-root `71f2a92292b86b46e374d2f9711f63d89bd8324f`
* tinygo `b5a8931fb57ca93fdfcf9c4e443833cd0def2a8d`

It looks like your system might not have the library `libpthread` installed. Afaik you can verify this with the following command:

```
ldconfig -p | grep pthread
```

You have to install the library if it does not show any output.

jensdrenhaus, 2024-12-06:
@leongross and I encounter the same problem with recent commits. We definitely had a working state one, but cannot reproduce that working state, there must have been some regression introduces. #3238 will help to avoid those regressions. 
#3239 will fix the actual current situation.

closed 2024-12-09 by leongross