u-root/u-root #3231: tinygo `ntpdate` does not return date properly

10000TB, 2024-12-03:
**Describe the bug**
A clear and concise description of what the bug is.

tinygo `ntpdate` does not return date properly

```
[root/cmds/core/ntpdate$](https://www.google.com/url?q=http://github.com/u-root/u-root/cmds/core/ntpdate$&sa=D&source=docs&ust=1733258579673867&usg=AOvVaw3flOcTpyo3yqbU8yGtphLj) ./ntpdate.tiny -verbose
2024/12/03 19:33:33 Reading NTP servers from config file: /etc/ntp.conf
2024/12/03 19:33:33 Unable to open config file: open /etc/ntp.conf: file does not exist
2024/12/03 19:33:33 No servers provided, falling back to [time.google.com](https://www.google.com/url?q=http://time.google.com&sa=D&source=docs&ust=1733258579674083&usg=AOvVaw1UmpDFABAcJRkOx2uLx5Y1)
2024/12/03 19:33:33 Getting time from [time.google.com](https://www.google.com/url?q=http://time.google.com&sa=D&source=docs&ust=1733258579674144&usg=AOvVaw0iWxMvzrpmX549ZUVUX_WR)
2024/12/03 19:33:33 Error getting time from [time.google.com](https://www.google.com/url?q=http://time.google.com&sa=D&source=docs&ust=1733258579674248&usg=AOvVaw0A4zs6wUPVl6pbHFThQLjA): Lookup of host name '[time.google.com](https://www.google.com/url?q=http://time.google.com&sa=D&source=docs&ust=1733258579674314&usg=AOvVaw3MRVgi-4GJCSbJy0QMbhOM)' failed: Netdev not set
2024/12/03 19:33:33 Error: unable to get time: unable to get any time from servers [[time.google.com](https://www.google.com/url?q=http://time.google.com&sa=D&source=docs&ust=1733258579674436&usg=AOvVaw0Y95aIqr73tHX_nExbW67s)]
```

**To Reproduce**
Steps to reproduce the behavior:

Checkout one of the recent commit like following,
```
xuehaohu@worldpeace2:~/go/src/github.com/u-root/u-root/cmds/core/ps$ git log -1
commit 69ad050bb861a6da0a69aa3399f938c01c8368de (HEAD -> main, origin/main, origin/HEAD)
Author: Jonathan Bluett-Duncan <jbluettduncan@gmail.com>
Date:   Wed Nov 27 23:29:14 2024 +0000

    cmds/extra/tsort: add benchmarking script

    Signed-off-by: Jonathan Bluett-Duncan <jbluettduncan@gmail.com>
```

build ntpdate with tinygo, and simply run it w/o any flags.

```
xuehaohu@worldpeace2:~/go/src/[github.com/u-root/u-root/cmds/core/ntpdate$] ls
ntpdate ntpdate.go ntpdate.tiny
xuehaohu@worldpeace2:~/go/src/[github.com/u-root/u-root/cmds/core/ntpdate$ ./ntpdate.tiny
2024/12/03 19:33:28 Error: unable to get time: unable to get any time from servers [time.google.com]
xuehaohu@worldpeace2:~/go/src/[github.com/u-root/u-root/cmds/core/ntpdate$] ./ntpdate.tiny -verbose
2024/12/03 19:33:33 Reading NTP servers from config file: /etc/ntp.conf
2024/12/03 19:33:33 Unable to open config file: open /etc/ntp.conf: file does not exist
2024/12/03 19:33:33 No servers provided, falling back to [time.google.com]
2024/12/03 19:33:33 Getting time from [time.google.com]
2024/12/03 19:33:33 Error getting time from [time.google.com]: Lookup of host name '[time.google.com]' failed: Netdev not set
2024/12/03 19:33:33 Error: unable to get time: unable to get any time from servers [time.google.com]
xuehaohu@worldpeace2:~/go/src/[github.com/u-root/u-root/cmds/core/ntpdate$ ./ntpdate
2024/12/03 19:33:37 Error: unable to set system time: operation not permitted
xuehaohu@worldpeace2:~/go/src/[github.com/u-root/u-root/cmds/core/ntpdate$ sudo ./ntpdate
2024/12/03 19:33:40 adjust time server [time.google.com] offset +0.000202 sec
xuehaohu@worldpeace2:~/go/src/[github.com/u-root/u-root/cmds/core/ntpdate$
```

**Expected behavior**
A clear and concise description of what you expected to happen.

(`ntpdate` used here was built with std golang toolchain)
```
xuehaohu@worldpeace2:~/go/src/[github.com/u-root/u-root/cmds/core/ntpdate$ ./ntpdate
2024/12/03 19:33:37 Error: unable to set system time: operation not permitted
xuehaohu@worldpeace2:~/go/src/[github.com/u-root/u-root/cmds/core/ntpdate$ sudo ./ntpdate
2024/12/03 19:33:40 adjust time server [time.google.com] offset +0.000202 sec
xuehaohu@worldpeace2:~/go/src/[github.com/u-root/u-root/cmds/core/ntpdate$
```

**Additional context**
Add any other context about the problem here.

jensdrenhaus, 2024-12-05:
On the journey towards tinygo integration into u-root the M3 milestone aims to make the cmds/core and cmds/exp commands buildable with the tinygo compiler. In some cases, there can still occur runtime errors.

However, we can take a look at this runtime problem on the way forward @leongross

jbduncan, 2024-12-10:
Should we include `cmds/extra` in tinygo milestone as well? :)