u-root/u-root #3232: tinygo `rmmod` does not build: `undefined: compressionReader`

10000TB, 2024-12-03:
**Describe the bug**
A clear and concise description of what the bug is.

```
xuehaohu@worldpeace2:~/go/src/github.com/u-root/u-root/cmds/core/rmmod$ $BASE/tinygo-org/tinygo/build/tinygo build -tags etgo,purego,noasm,tinygo.enable .
# github.com/u-root/u-root/pkg/kmodule
../../../pkg/kmodule/kmodule_linux.go:65:9: undefined: compressionReader
xuehaohu@worldpeace2:~/go/src/github.com/u-root/u-root/cmds/core/rmmod$
```

**To Reproduce**
Steps to reproduce the behavior:

```
xuehaohu@worldpeace2:~/go/src/github.com/u-root/u-root/cmds/core/rmmod$
xuehaohu@worldpeace2:~/go/src/github.com/u-root/u-root/cmds/core/rmmod$ ls
rmmod_linux.go
xuehaohu@worldpeace2:~/go/src/github.com/u-root/u-root/cmds/core/rmmod$ $BASE/tinygo-org/tinygo/build/tinygo build -tags etgo,purego,noasm,tinygo.enable .
# github.com/u-root/u-root/pkg/kmodule
../../../pkg/kmodule/kmodule_linux.go:65:9: undefined: compressionReader
xuehaohu@worldpeace2:~/go/src/github.com/u-root/u-root/cmds/core/rmmod$
xuehaohu@worldpeace2:~/go/src/github.com/u-root/u-root/cmds/core/rmmod$ git log -1
commit 69ad050bb861a6da0a69aa3399f938c01c8368de (HEAD -> main, origin/main, origin/HEAD)
Author: Jonathan Bluett-Duncan <jbluettduncan@gmail.com>
Date:   Wed Nov 27 23:29:14 2024 +0000

    cmds/extra/tsort: add benchmarking script

    Signed-off-by: Jonathan Bluett-Duncan <jbluettduncan@gmail.com>
xuehaohu@worldpeace2:~/go/src/github.com/u-root/u-root/cmds/core/rmmod$
```

**Expected behavior**
A clear and concise description of what you expected to happen.

**Additional context**
Add any other context about the problem here.

leongross, 2024-12-05:
Pull Request #3235 should have resolved this issue. That means commit 439a6abc142872bc0fe09e6299376cacec2c292f should built just fine.

jensdrenhaus, 2024-12-05:
Works for my with more recent commit:
```

closed 2024-12-05 by jensdrenhaus