u-root/u-root #3258: Reproducibility of pinned go modules since switch from vendoring for a commit in the future?

tlaurion, 2024-12-27:
> @hugelgupf I do see any pinning for random example https://github.com/u-root/u-root/blob/main/pkg/boot/kexec/kexec_load_linux.go#L12
> 
> I get that at a moment in time, if one builds the same initramfs twice it will pass the test. My point, irrelevant to subject of this ticket is if someone wants to rebuild same commit in 6 months it won't produce the same initramfs because dependencies might evolve as well since not pinned to dependencies commit, simply downloading latest available version at moment of building.

I see what you mean -- though that's a Go problem with Minimal Version Selection in Go modules. I hadn't realized that Go lost that with the switch from vendoring to modules. (We still have the vendor directory, so theoretically u-root hasn't lost it if you build with modules disabled.)

_Originally posted by @hugelgupf in https://github.com/u-root/u-root/issues/2687#issuecomment-1567480681_

tlaurion, 2025-01-02:
@rminnich : any input?

orangecms, 2025-11-12:
I have just taken a look at the `go.sum` file, which _does_ contain hashes, but:

- hashes are for dependencies themselves and many other `go.mod`  files, though notably, _not_ their `go.sum` files
- dependencies are not guaranteed to pin _their_ dependencies, since they are outside u-root's control

AFAIUI, reproducibility is possibly not the goal of Go modules, but rather _freshness_.