u-root/u-root #3267: Inconsistent command-line verbose logging flags

pamolloy, 2025-01-08:
`boot -v` and `localboot -d` print verbose debug logging. It would be nice if they (and perhaps other commands) used the same flag to enable debug logging.

leongross, 2025-04-10:
I think the general idea here is to be compatible with the original GNU coreutils, iputils and so on; especially in the user interface, so also the way the flags are passed. This eliminates porting work for scripts, etc. If there are inconsistencies with the original implementations, feel free to point them out so we can align them.

pamolloy, 2025-04-10:
Thanks for the reply @leongross. I'm sorry, but I don't entirely follow. Are there are `boot` and `localboot` implementations in coreutils, etc? I thought they were unique to u-root. How is the fact that `boot` and `localboot` seemingly using different flags for the same thing dependent on coreutils, etc?