u-root/u-root #3361: Proposal: ping: raise default wait time

quite, 2025-04-23:
uroot ping command's current "wait time" is a very short 100 milliseconds. For my use-cases, I always have to pass approximately `-w 250` when testing connectivity using ping (and less often 5 or 10 times that value).

The broader issue is that ping really does not behave at all like the typical (iputils) ping command. From what I understand, iputils-ping has a timeout of 10 seconds when there is no answer, and this can be adjusted using its `-W` (uppercase) option. If there is an answer, the timeout for the next answer is 2*RoundTripTime.

uroot-ping takes the flag `-w` to adjust the default timeout. This timeout (variables `wtf` and `waitFor` in the code) is a deadline which spans over both writing the ICMP echo packet, and reading up a (reply) packet.

I would suggests that the default "wait time" for our ping is raised at least 10 times to 1000 milliseconds (1 second). Also that the very different and simplified behaviour of our ping cmd, compared to iputils-ping, is documented briefly in the command's help/usage.

orangecms, 2026-02-18:
Looks like Windows uses a 4000ms default and lower-case `/w`, just for comparison:
https://learn.microsoft.com/en-us/windows-server/administration/windows-commands/ping

FreeBSD also uses `-W`, like iputils, whatever the default may be:
https://man.freebsd.org/cgi/man.cgi?ping(8)

orangecms, 2026-02-18:
per https://github.com/u-root/u-root/pull/3510#pullrequestreview-3821279824 we just agreed on that one second 🙂