u-root/u-root #3362: gosh with goshliner, program cannot read Stdin (also Ctrl-c to same program exits gosh)

quite, 2025-04-23:
A Go program that does `bufio.NewReader(os.Stdin).ReadString('\n')` does not get any input when run in the gosh shell that is built with `goshliner` tag. It works fine in the regular gosh that uses `bubbline`.

With the `tst` program built from the code further down, I run u-root like this:

```
./u-root -tags goshliner -files ../src/tst:bin/tst -o initramfs "./cmds/core/*"
```

Then extracting the initramfs cpio, and running `bin/defaultsh`, and from inside that `bin/tst`. Nothing happens when hitting `Enter`.

When reproducing this again now, I also notice that when in gosh-goshliner, running `tst` and hitting `Ctrl-c` exits the whole gosh shell. But when running gosh-bubbline, hitting `Ctrl-c` returns me to gosh, as expected.

Minimal ReadString tst program:

```
package main

import (
    "bufio"
    "fmt"
    "os"
)

func main() {
    fmt.Print("Commit by hitting Enter key, cancel with Ctrl-c")
    r := bufio.NewReader(os.Stdin)
    _, err := r.ReadString('\n')
    if err != nil {
        fmt.Printf("read failed: %s\n", err)
        os.Exit(1)
    }
    fmt.Printf("Will proceed!\n")
}
```

MDr164, 2025-06-10:
A similar issue exists with the tag goshsmall and both goshliner and goshsmall are getting more important so I'm trying to get this resolved asap.

MDr164, 2026-05-04:
@quite can you test the code from https://github.com/u-root/u-root/pull/3586 in your environment? It seems to have solved it for me but my test environment seems to not reflect most others for some reason most of the time